mvp/uhubctl #366: Cannot power cycle non-responsive Unifying Receiver

alanzchen, 2021-10-14:
I have a strange case: both my old and a brand new Logitech Unifying Receiver would stop responding (partially). The mouse and keyboard still work, but Logi Options (the software that provides higher level functions such as custom buttons, etc) stops working. Unplugging and replugging the Unifying Receiver would resolve this issue immediately. So I stumble upon this project and hope it can automate the task for me.

I am using a M1 MacBook Air with Big Sur, connected with a Thunderbolt 3 Dock (HP ZBook 150W Thunderbolt 3 Docking Station P5Q58UT#ABA)

Here are some clues of what may be happening:
 - All other USB ports / devices work normally while Unifying Receiver becomes partially unresponsive.
 - While it is unresponsive, running `uhubctl` would print other hubs and ports normally until it reaches the port where the Unifying Receiver is. I have to ctrl + C to abort.
 - Similarly, trying to power cycle (e.g., `uhubctl -e -l 2-3 -p 2 -a cycle -d 0`) cause the same hanging issue.
 - I am able to power cycle that port and the Unifying Receiver when it is still fully responsive.
 - When the Unifying Receiver is unresponsive, I am able to recognize it in macOS’s "System Report”, “USB” page.

Sample output for `uhubctl`:
```
Current status for hub 2-3 [174c:2074 Asmedia ASM107x 0000000A0003, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
^C
```

And the output right after I replugged the Unifying Receiver:

```
Current status for hub 2-3 [174c:2074 Asmedia ASM107x 0000000A0003, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0103 power enable connect [046d:c52b Logitech USB Receiver]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 2-1 [174c:3074 Asmedia ASM107x 0000000A0003, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 3-3.4 [043e:9a71, USB 3.00, 3 ports, ppps]
  Port 1: 06a0 power Rx.Detect
  Port 2: 06a0 power Rx.Detect
  Port 3: 0203 power 5gbps U0 enable connect [043e:9a68 LG Electronlcs Inc. LG UltraFine Display Camera]
Current status for hub 3-2.4 [043e:9a73 1D0218512AE2, USB 2.10, 3 ports, ppps]
  Port 1: 0503 power highspeed enable connect [043e:9a66 LG Electronics Inc. LG UltraFine Display Audio]
  Port 2: 0103 power enable connect [043e:9a70 LG Electronics Inc. LG UltraFine Display Controls]
  Port 3: 0100 power
Current status for hub 3-3 [043e:9a60 LG Electronics Inc. USB3.1 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [0781:558c SanDisk Extreme SSD 313932383656343030333435]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect [043e:9a71, USB 3.00, 3 ports, ppps]
Current status for hub 3-2 [043e:9a61 LG Electronics Inc. USB2.1 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [043e:9a73 1D0218512AE2, USB 2.10, 3 ports, ppps]
```

Any thoughts?

mvp, 2021-10-14:
uhubctl is using libusb calls to query list of USB devices and their status, as well as changing power state. libusb, in turn, has set of primitives to achieve these tasks (implemented differently on each OS, e.g. MacOS, Linux, etc). Most of these primitives are calling into kernel and appropriate USB drivers. In is rare, but not impossible that kernel USB driver (or driver for particular USB device type) has a bug in it, and becomes unresponsive. In this case, you can't really do much of anything except to ask OS vendor (Apple) to fix underlying kernel bug. However, before jumping to conclusions, I recommend to upgrade MacOS (if you haven't upgraded yet), as well as use latest version of libusb (1.0.24 as of this writing) - it is possible that they have a fix for this. Also, as far as I know, Logitech unifying receiver itself has own updatable firmware - it might be good idea to update it too.

At any rate, if even trying to list devices hangs, it is not uhubctl problem - all that does is to call libusb function to get list of known USB devices, which in turn calls OS primitive to enumerate them.

alanzchen, 2021-10-14:
> I recommend to upgrade MacOS (if you haven't upgraded yet), as well as use latest version of libusb (1.0.24 as of this writing)

I am using 11.6 (20G165), which is the current latest stable version. My libusb is at 1.0.24. The Unifying Receiver has 24.11 (36), which is the latest firmware.

> At any rate, if even trying to list devices hangs, it is not uhubctl problem

Yes, that’s what I’d expect. Thanks for developing this project and it is fun to play with it.

It is possible to bypass querying the USB status and try to directly change the power state? Currently it didn’t even try to send signals to power cycles the port because it hangs at querying.

BTW, you may add the hubs listed in my outputs to compatible hubs in README.

mvp, 2021-10-14:
Theoretically it might be possible to bypass querying, but it will require patching uhubctl source. And, it is very likely to not actually help - but feel free to give it a try.

Regarding adding new hubs - thanks for the offer, but unfortunately, just looking at uhubctl output is not enough to uniquely identify them. All too often, completely different hubs use identical USB chips (including their USB descriptor strings), but use different circuitry or configure USB chip differently. For that reason we request to give more information about supported hubs, ideally model name printed on it or link to any site where it can be purchased.

mvp, 2021-10-29:
This issue should be fixed by f8cb89f4.