mvp/uhubctl #373: CM4 + LAN9514 Not detected

synapsertu, 2021-11-09:
Hi

I have used this program successfully in the past on a CM3 based unit (see #90) but I can't for the life of me work out what I'm not doing right on a new CM4 based unit

This is the setup 
```
root@raspberrypi:~/uhubctl-master# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/0p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 3: Dev 4, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
        |__ Port 4: Dev 5, If 1, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 4: Dev 5, If 4, Class=Vendor Specific Class, Driver=qmi_wwan, 480M
        |__ Port 4: Dev 5, If 2, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 4: Dev 5, If 3, Class=Vendor Specific Class, Driver=option, 480M
        |__ Port 5: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M

root@raspberrypi:~/uhubctl-master# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0424:2240 Standard Microsystems Corp.
Bus 001 Device 005: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 001 Device 004: ID 413c:2003 Dell Computer Corp. Keyboard
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

Note I'm using the below in config.txt to enable the "correct" USB controller as the dwc2 interface is buggy for highspeed transfers

`otg_mode=1`

Running the program gives

```
root@raspberrypi:~/uhubctl-master# ./uhubctl
No compatible devices detected!
Run with -h to get usage info.
```

Even if I try and coax it into recognising the LAN9514 chip I can't get any response

```
root@raspberrypi:~/uhubctl-master# dmesg |grep 0424
[    2.003814] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 1.00
[    2.578301] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 1.00
[    4.070361] usb 1-1.5: New USB device found, idVendor=0424, idProduct=2240, bcdDevice= 1.98

root@raspberrypi:~/uhubctl-master# ./uhubctl -L 1-1
No compatible devices detected!
Run with -h to get usage info.

root@raspberrypi:~/uhubctl-master# ./uhubctl -n 0424
No compatible devices detected!
Run with -h to get usage info.
```

Any ideas on what I can try here?

Thanks!

Andrew.

mvp, 2021-11-17:
I assume you are using most recent uhubctl version (2.4.0 as of this writing).
If uhubctl does not show anything running as root, it means your devices do not advertise power switching in device descriptors. You can try forcing it with `-f` option, but it is not likely to help.

closed 2021-11-17 by mvp

synapsertu, 2021-11-17:
> it means your devices do not advertise power switching in device descriptors.
The LAN951x range is indeed capable of this function, see #90 

When you say "do not advertise power switching in device descriptors"

What are you looking for and can I use udevadm or similar to dump these to check?

Thanks!

synapsertu, 2021-11-17:
Yes that worked with the -f thing added -- see below

```
root@raspberrypi:~/uhubctl-master# ./uhubctl -f
Current status for hub 2 [1d6b:0003 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller fe9c0000.xhci, USB 3.00, 0 ports, nops]
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports, ganged]
  Port 1: 0503 power highspeed enable connect [0424:ec00 SMSC LAN9514 00951400]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2c7c:0125 Android Android]
  Port 5: 0503 power highspeed enable connect [0424:2240 Generic Ultra Fast Media 000000225001]
Current status for hub 1 [1d6b:0002 Linux 5.10.17-v7l+ xhci-hcd xHCI Host Controller fe9c0000.xhci, USB 2.00, 1 ports, nops]
  Port 1: 0503 power highspeed enable connect [0424:9514, USB 2.00, 5 ports, ganged]

root@raspberrypi:~/uhubctl-master# ./uhubctl -f -l 1-1
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports, ganged]
  Port 1: 0503 power highspeed enable connect [0424:ec00 SMSC LAN9514 00951400]
  Port 2: 0100 power
  Port 3: 0303 power lowspeed enable connect [413c:2003 Dell Dell USB Keyboard]
  Port 4: 0503 power highspeed enable connect [2c7c:0125 Android Android]
  Port 5: 0503 power highspeed enable connect [0424:2240 Generic Ultra Fast Media 000000225001]

root@raspberrypi:~/uhubctl-master# ./uhubctl -l 1-1
No compatible devices detected at location 1-1!
Run with -h to get usage info.

root@raspberrypi:~/uhubctl-master# ./uhubctl -f -l 1-1 -a0 -p3 -r 100
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports, ganged]
  Port 3: 0303 power lowspeed enable connect [413c:2003 Dell Dell USB Keyboard]
Sent power off request
New status for hub 1-1 [0424:9514, USB 2.00, 5 ports, ganged]
  Port 3: 0000 off

root@raspberrypi:~/uhubctl-master# lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0424:2240 Standard Microsystems Corp.
Bus 001 Device 004: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```