mvp/uhubctl #376: Additional supported hubs

Tbruno25, 2021-11-22:
Startech ‎ST4300USB3 
Startech ‎ST7300USB3B 
Anker AK-68ANHUB-BV7A-0004

I verified VBUS can be controlled on all ports for the above 3 hubs.

ps -- thank you for this tool! =]

bigrck64, 2021-11-23:
Thanks for this input, it's very hard to find compatible hub nowadays :)
I just bought one Startech ST7300USB3B to check it :+1:

Tbruno25, 2021-11-23:
> Thanks for this input, it's very hard to find compatible hub nowadays :) I just bought one Startech ST7300USB3B to check it +1

@bigrck64 I unfortunately made a mistake when submitting this.
The Startech 4 port hub and Anker 7 port are functional, but the 7 is not.

bigrck64, 2021-11-24:
Arg to late, I will receive it in 5 days !
Well, I will be able to confirm the ST7300USB3B is not working before sending it back to amazon :)
I'm a bit sad because it was the only one with a 12V x 3.0A power block.

mvp, 2021-11-29:
Thanks for you report, AK-68ANHUB-BV7A-0004 was already in supported table.
Adding StarTech ST4300USB3

closed 2021-11-29 by mvp

Tbruno25, 2021-12-02:
The ST7300USB3B is indeed functional as I first noted.

Tbruno25, 2021-12-02:
@mvp Something is strange and I get different behavior depending on the host pc.

I tested using 2 different laptops: 
1. Thinkpad T14 
2. Thinkpad L380

Both are running Ubuntu 20.04 with uhubctl compiled from source.

ST7300USB3B
1. None of the ports work
2. All 7 ports work.

AK-68ANHUB-BV7A-0004
1. Only 1 port works.
2. All 7 ports work.

Maybe this relates to https://github.com/mvp/uhubctl/issues/371

mvp, 2021-12-02:
@Tbruno25 , hub operation should not depend on host type or host OS, only on hub hardware.

How are you testing if turning power off actually is working?
I recommend following:
1. Ensure both hosts have identical USB cables and the same connection type (USB3 preferably, but both USB2 is ok too). If uhubctl shows 2 hub instances (e.g. USB 2 and USB 3) it means it's connected as USB 3. Otherwise it will be USB 2.
2. When powering off, ensure that for USB 3 connection uhubctl is sending power off for both USB2 and USB3 components, or it will not work correctly. Latest version 2.4.0 should handle this properly, but it is possible that uhubctl gets confused and doesn't detect USB 2/USB 3 duality properly - please post output of your uhubctl commands here.
3. To check if power really cut off, you should ideally measure voltage on USB pins. But usually, checking if connected phone stops charging or USB lamp stops shining is enough.
4. You maybe hitting Linux kernel issue when it tries to automatically turn power back on for devices which were turned off by uhubctl (there is FAQ entry about it). Usually, you can brute force it with -r 1000 option.

jclulow, 2021-12-09:
I just bought a Startech ST4300USB3 and it does not seem to work?  It seems like `uhubctl` is trying to do the right thing with both (USB 2/3) virtual hubs, but the USB light I have plugged in does not turn off.

```
$ sudo ./uhubctl -l 2-1 -p 1-4 -a off
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Sent power off request
New status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off

$ sudo ./uhubctl
Current status for hub 2-1 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
```

Using `-r 1000` does not seem to change things.

```
$ head -1 /etc/os-release ; uname -a
PRETTY_NAME="Ubuntu 21.10"
Linux leftstation 5.13.0-22-generic #22-Ubuntu SMP Fri Nov 5 13:21:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

jclulow, 2021-12-09:
@Tbruno25 What does the barcode on the base of your ST4300USB3 say?  There is a serial number, in bold print, but underneath is what looks like a batch or version number:

`U3-V20-UH24-14-02361`

sjmtlewy, 2022-07-22:
Just tried ST7300USB3B - does not disconnect VBUS

commit 610e85ff (2021-11-29): Add StarTech ST4300USB3 as compatible device

Closes #376.