mvp/uhubctl #378: Can only control two charging ports on Plugable USB3-HUB7-81X

mtfurlan, 2021-12-06:
I have a Plugable USB3-HUB7-81X and I can only control power on the two charging ports, same as the Plugable USB3-HUB7C.

Do you want me to email plugable and try to see if they ever changed it, or do we just assume all USB3-HUB7-81X only work on the charging ports?

mvp, 2021-12-07:
Well, it does not hurt to ask :-) Certainly, ask them please!

mtfurlan, 2021-12-07:
> To my knowledge there was no change during the manufacturing of that USB HUB. Typically when a change is needed for the electronics from a device, we will do a relaunch as a new product so that the support for our products remains clear.

They suggested the [usbc-hub7bc](https://plugable.com/products/usbc-hub7bc) as the only plugable USB hub that had full PPPS, and that is already in the list.

bigrck64, 2021-12-10:
For me,  usbc-hub7bc is not referenced into to uhubctl readme, there is only the USB3-HUB7BC.
https://github.com/mvp/uhubctl

Did you have one usbc-hub7bc in stock to test it ? this model is pretty recent (2021.07)

mtfurlan, 2021-12-13:
I now have a USBC-HUB7BC, and 6/7 of the ports work.
I've opened another ticket with them to ask if that's expected.

Port numbering left to right:
* 3.3
* 3.2
* 3.1
* 3.4
* 2
* 1
* 4(no PPPS)

bigrck64, 2021-12-17:
I also received my Plugable USBC-HUB7BC today, and I can confirm 6 or 7 ports are working with uhubctl !
@mvp : you can add it to readme :+1: 

port1 = 3.3
port2 = 3.2
port3 = 3.1
port4 = 3.4
port5 = 2
port6 = 1
port7 = no PPPS

PR #379 (2021-12-17): Add Plugable USBC-HUB7BC, note USB3-HUB7-81x is 2 port only
Closes #378.
commit 61187d75 (2021-12-17): Add pluggable USB3-HUB7BC; note USB3-HUB7-81x is 2 port only

Closes #378.
closed 2021-12-17 by mvp