mvp/uhubctl #382: Multiple Calls Fail....

MattDralle, 2022-01-04:
Hello,

I changed your main(), to the USBPortPower() as shown below. So that I could integrate the functionality into another program.  But, I've ran into a problem.  The first time I call  USBPortPower() with the options ' "cycle" "1-1" "3" ".5" "20" "1" "" "" "no" ' It works great, Port 3 is power cycled as requested.  BUT, the second time I call it exactly the same way, I get the error "changing port state for multiple (2) hubs at once is not supported."  I've tried a variety or location options, but I can't get it to work more than one time.  Killing my program and starting it again will yield another successful power cycle.  

Please help... -Matt Dralle

```
**int USBPortPower( char * Action, char * Location, char * Ports, char * Delay, char * Wait, char * Repeat, char * Search, char * Vendor, char * Reset )
{
int rc;
int c = 0;
int option_index = 0;

	strncpy( opt_location, Location, sizeof( opt_location ) );
	strncpy( opt_vendor, Vendor, sizeof( Vendor ) );
	strncpy( opt_search, Search, sizeof( Search ) );
	opt_ports = ports2bitmap( Ports );
	if( !strcasecmp( Action, "off" )   || !strcasecmp( Action, "0" ) ) opt_action = POWER_OFF;
	else if( !strcasecmp( Action, "on" )    || !strcasecmp( Action, "1" ) ) opt_action = POWER_ON;
	else if( !strcasecmp( Action, "cycle" ) || !strcasecmp( Action, "2" ) ) opt_action = POWER_CYCLE;
	else if( !strcasecmp( Action, "toggle" ) || !strcasecmp( Action, "3" ) ) opt_action = POWER_TOGGLE;
	opt_delay = atof( Delay );
	opt_repeat = atoi( Repeat );
	opt_nodesc = 0;
	opt_exact = 0;
	opt_reset = ( !strcasecmp( Reset, "yes" ) || !strcasecmp( Reset, "1" ) )?1:0;
	opt_wait = atoi( Wait );
	farprintfDebugInfoHeader( DEBUG );
	printf( "%sUSB Port Power%s: %s\"%s%s%s\" USB Port( s ): \"%s%s%s\": Beginning...%s\n", 
			LIGHT_PURPLE, 
			YELLOW, 
			LIGHT_GREEN, 
			LIGHT_CYAN,
			opt_action==POWER_OFF?"Turning OFF":\
			opt_action==POWER_ON?"Turning ON": \
			opt_action==POWER_CYCLE?"Power Cycling":\
			opt_action==POWER_TOGGLE?"Power Toggling":\
			"Unknown",
			LIGHT_GREEN, 
			LIGHT_CYAN, 
			Ports, 
			LIGHT_GREEN, 
			NORM );**

```

mvp, 2022-01-04:
I don't understand what problem you are trying to solve.
I don't recommend changing uhubctl code, instead just call `uhubctl -l location -p ports -a action` as external process to your program.

 The only way to get this error
```
changing port state for multiple (2) hubs at once is not supported
```
is if you are calling uhubctl without `-l` option.
Ideally, you should always use `-l location`, `-p ports` and `-a action`.
* location is hub id, e.g. `1-1`. if location is omitted, it will try changing state for supported hub, but if there is more than one, it will fail giving you error above
* ports is list of ports to change state for. if omitted, defaults to all ports of given hub.
* action is one of `on`, `off`, `cycle` or `toggle`. if omitted, simply prints current ports status.

MattDralle, 2022-01-04:
Thank you for the reply.  I prefer to not use a system() call in my app.  I am using effectively the -l 1-1 option on both calls.  But the second time I call it, without restarting the code, I get the "multiple hubs" error.  On the second call, usb_find_hubs() is setting hub_phys_count = 2, compared to the first call where it is returning "1".  Both calls effectively spec -l as "1-1" and -p as "3".

I used the options:     "cycle" "1-1" "3" ".5" "20" "1" "" "" "no" 
Location = 1-1
Action = cycle
Ports = 3

[USBControl.zip](https://github.com/mvp/uhubctl/files/7810783/USBControl.zip)

mvp, 2022-01-05:
One more reason why you should not change uhubctl source code is GPL license. unless you are willing to put your new code under GPL, you should not do it. Calling uhubctl is fine.

To fix your issue, I guess you can simply re-initialize all static variables used by uhubctl beween your function invocations. Current code expects to be called from command line only.

mvp, 2022-01-16:
Closing for inactivity. Again, be careful to copy/paste uhubctl source code as part of your program unless you intend to license your new code as GPL too - otherwise it will be license infringement. Please reopen if you have anything to add.

closed 2022-01-16 by mvp

MattDralle, 2022-01-17:
I never could get it to work reliably on multiple invocations, so I dumped it.  Thanks for your time. -Matt