mvp/uhubctl #384: Add comptabile USB hub

acosonic, 2022-01-19:
Lenovo ThinkVision displays come with 4 port hub, where one of ports is yellow and has battery icon on it... 

That's supposed to be used for charging phones etc...

Device name is:
[17ef:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports]

lsusb says:
Bus 001 Device 006: ID 17ef:0610 Lenovo USB2.1 Hub

Bus 001 Device 006: ID 17ef:0610 Lenovo USB2.1 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x17ef Lenovo
  idProduct          0x0610 
  bcdDevice           60.32
  iManufacturer           1 GenesysLogic
  iProduct                2 USB2.1 Hub
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Single TT
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 TT per port
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00e9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 32 FS bits
    Port indicators
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0503 highspeed power enable connect
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x002a
  bNumDeviceCaps          3
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat           8 micro seconds
    bU2DevExitLat         190 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {74b951c7-5a96-844a-b874-ceeca2d8868e}
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

mvp, 2022-01-19:
Thanks for your report! Can you please check if VBUS is turning off? That is, connected phone stops charging when you turn power off with uhubctl?

acosonic, 2022-01-19:
Yup I have tested it, that's why I am reporting... But I am seeing for a way to charge phone faster... Any idea how to achieve that...?

mvp, 2022-01-19:
Afaik, the only way to charge phone faster is if your both USB port and your phone support fast charging protocols, such that QC (Qualcomm Quick Charge), AFC (Samsung Adaptive Fast Charging) or PD (Power Delivery). All of those work by increasing either current or voltage beyond standard 5V/0.5A that USB is guaranteeing. e.g they can increase current to 2A-3A (up to 5A), and voltage to 9V (up to 20V), so theoretical charger max is 100W (=5A*20V), however most phones max out at 18W-30W max charge speed. Those fast charging protocols require support from both charger and your client device - they will only engage if both sides agree to switch to faster charge speed. Some (more expensive) USB hubs feature dedicated charge ports with QC or PD support, you should be looking for those.

mvp, 2022-01-19:
Regarding Lenovo ThinkVision display - @acosonic  can you please provide actual model you tested to work? We do not add wide classes, as quite often it does not work for all models. Thanks!

acosonic, 2022-01-20:
ThinkVision model:
LEN T24i-10

So seems that all ports work...

`uhubctl 
Current status for hub 1-11 [17ef:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [18d1:4ee7 Xiaomi Mi 9T b0664b57]
  Port 4: 0100 power
root@acop:/home/acop# uhubctl -a off -p 3
Current status for hub 1-11 [17ef:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports]
  Port 3: 0503 power highspeed enable connect [18d1:4ee7 Xiaomi Mi 9T b0664b57]
Sent power off request
New status for hub 1-11 [17ef:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports]
  Port 3: 0000 off
`

acosonic, 2022-01-20:
But it seems that xiaomi phone doesn't want to fast-charge via USB... Nor it takes more power from that yellow usb port with battery on it :( or any other port... It takes only up to 310 mA ...

acosonic, 2022-01-20:
Actually it's 3.1 hub, but needs to be plugged into proper port...

```
Current status for hub 2-5 [17ef:102a GenesysLogic USB3.1 Hub, USB 3.10, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-10 [17ef:0610 GenesysLogic USB2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power

```

openhardwarefan, 2022-01-21:
https://www.amazon.com/CableCreation-Multiport-Adapter-Gigabit-Ethernet/dp/B08FWMWGTD/ref=sr_1_3?keywords=cablecreation%2Busb%2Bc%2Bhub&qid=1642792024&sprefix=cablecrea%2Caps%2C116&sr=8-3&th=1
To power off the usb port right next to Micro SD card slot:
uhubctl -e -l 2-1 -p3 -a 0
uhubctl -e -l 1-3 -p3 -a 0

To power off the other usb port close to Ethernet port:
uhubctl -e -l 2-1 -p3 -a 0
uhubctl -e -l 1-3 -p2 -a 0

mvp, 2022-02-02:
@acosonic, sorry can't just add Lenovo ThinkVision to the list, need full model number, ideally link to manufacturer site or amazon page

closed 2022-02-02 by mvp

acosonic, 2022-02-02:
I gave it already... Above: [https://github.com/mvp/uhubctl/issues/384#issuecomment-1017328007](https://github.com/mvp/uhubctl/issues/384#issuecomment-1017328007)

ThinkVision model:
LEN T24i-10

Manufacturer site:
[https://www.lenovo.com/ao/en/monitors/t24i](https://www.lenovo.com/ao/en/monitors/t24i
)
_

mvp, 2022-02-02:
Oops, sorry. I guess I'm blind. Or reading it diagonally :-)

acosonic, 2022-02-10:
I can confirm LEN T24i-20 also works, https://support.lenovo.com/rs/en/solutions/pd500403-thinkvision-t24i-20-61f7-mar1-ww-monitor-overview it's the same USB hub.

commit ed27158c (2022-02-03): Add Lenovo T24i-10 as compatible device

Closes #384.