mvp/uhubctl #388: Add compatible hub

bastianh, 2022-01-20:
I was looking for an usb hub with a better power supply that could power my raspi and also allows switching off the backlight of the connected lcd panel so I was trying the [ICY BOX IB-HUB1703-QC3](https://www.amazon.de/ICY-BOX-IB-HUB1703-QC3-Industrie-Montage-Halterung/dp/B07CCK5RK3/ref=sr_1_2_sspa)

I tried with a connected phone and the 7 blue usb ports are switchable with uhubctl

    Current status for hub 2-1.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
      Port 1: 02a0 power 5gbps Rx.Detect
      Port 2: 02a0 power 5gbps Rx.Detect
      Port 3: 02a0 power 5gbps Rx.Detect
      Port 4: 02a0 power 5gbps Rx.Detect
    Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
      Port 1: 02a0 power 5gbps Rx.Detect
      Port 2: 02a0 power 5gbps Rx.Detect
      Port 3: 02a0 power 5gbps Rx.Detect
      Port 4: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
    Current status for hub 1-1.1.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
      Port 1: 0100 power
      Port 2: 0100 power
      Port 3: 0100 power
      Port 4: 0100 power
    Current status for hub 1-1.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
      Port 1: 0100 power
      Port 2: 0100 power
      Port 3: 0100 power
      Port 4: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]

bastianh, 2022-01-20:
I wonder if the device maybe has a problem with switching? I try to turn off a lcd display that is connected via hdmi for image and usb for touch and power... it looks like it enables itself again 

    pibone# uhubctl -l 1-1.1 -p 2 -a off
    Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
      Port 2: 02a0 power 5gbps Rx.Detect
    Sent power off request
    New status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
      Port 2: 00a0 off
    Current status for hub 1-1.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
      Port 2: 0103 power enable connect [0eef:0005 BYZHYYZHY By ZH851 @?P39116642]
    Sent power off request
    New status for hub 1-1.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
      Port 2: 0000 off
    
    pibone# uhubctl -l 1-1.1 -p 2
    Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
      Port 2: 00a0 off
    Current status for hub 1-1.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
      Port 2: 0000 off
    
    pibone# uhubctl -l 1-1.1 -p 2
    Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
      Port 2: 00a0 off
    Current status for hub 1-1.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
      Port 2: 0101 power connect [0eef:0005]
    
    pibone# uhubctl -l 1-1.1 -p 2
    Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
      Port 2: 00a0 off
    Current status for hub 1-1.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
      Port 2: 0103 power enable connect []
    
    pibone# uhubctl -l 1-1.1 -p 2
    Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
      Port 2: 00a0 off
    Current status for hub 1-1.1 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
      Port 2: 0103 power enable connect [0eef:0005 BYZHYYZHY By ZH851 @?P39116642]

I have no issues turning off the phone .. but turning of an usb stick or drive also has no effect?

mvp, 2022-01-20:
First, make sure you use uhubctl 2.4.0 or later.
This might be this issue https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux
Can you try adding `-r 1000` to your command line?
Also, it is worth checking `dmesg` logs to see what is turning power back on for USB3.

bastianh, 2022-01-20:
I'm using `2.4.0-24-g61187d75`

with the `-r 1000` the port stays off in uhubctl, and the touch function is not working anymore .. but the display backlight is still on :)

I retestet turning on and off charging with an iphone on the same port and it works as expected.. on, off, cycle .)
Is it possible that the backlight is getting power from the port even if turned off? I don't really have other usb gadgets to test here.

mvp, 2022-01-20:
One possiblity is that this hub still has trickle voltage even if port is off, which is enough to have your backlight still work, but not enough to phone or other USB devices to recognize it as working. Best way to answer this is with voltmeter.

bigrck64, 2022-01-24:
Hey Bastianh, thanks for report, I just bought one IB-HUB1703-QC to test it.
@mvp: I will check all ports and report if vbus is really off in a few days :)

bigrck64, 2022-01-27:
Well, I receive the IB-HUB1703-QC3 this morning (serial= 6036221390282), and I could not control any of the blue ports :(
- I'm using uhubctl 2.4.0-24-g61187d75 with Linux version 5.4.157-1-pve
- I tested the hub connected on USB2 and USB3 pc port
- I tested with -r 1000 on all ports
- hub is well announced as PPPS, but my USB flashlight does NOT turn off

For me it's a no go :(

I feel the HUB is activated when you turn on the ON/OFF physical button, and after that, we got no control over it.
@bastianh: you said you could turn off one phone, what phone model are you speaking ? and how did you check power was off ? and what was the connected blue port on usb hub ?

Current status for hub 2-2.4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
Current status for hub 1-2.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-2 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]

bigrck64, 2022-01-27:
OK, I connected one iPhone 6s there, and you were right, I can see the usb charging sign OFF or ON when I'm controling the port with uhubctl :+1: 

So this HUB is partially working: 
- it correctly shutting down enough power to stop charging one iPhone
- but it left enough power to make one usb light working

bigrck64, 2022-01-27:
More tests show me that iPhone is effectively no more displaying the "Charging" icon when the port is "off", but in real it's  correctly receiving enough power to slowly charge the battery.
So... for me, we can't really control the power on blue usb ports, sorry :(

@bastianh : can you try other usb device than phone and see if you can control the power for they, you speak about a raspi, can you turn it off on usb blue port ?

bastianh, 2022-01-27:
I can't really test it with the pi, because it's the controller of the hub .. but I also tested with hard drives after unmounting and they are not powering off completly :(

mvp, 2022-01-27:
Note that for hard drives, it may be necessary to use -r 1000, as some devices are surprised by power removal and try to automatically turn power back on. Please read this FAQ: https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux

mvp, 2022-02-02:
Closing as this hub does not seem to support vbus off correctly.

closed 2022-02-02 by mvp