mvp/uhubctl #390: StarTech	ST4300USB3 not compatible

scottagould, 2022-01-24:
After searching for a compatible USB Hub, the best I could do was find a StarTech ST4300USB3

However upon testing, it doesn't actually cut power, only data. It's the same as using a cheap $7 USB hub and the `-f` command

You may want to remove this from your compatible device list. 

As for me, my quest to source a different hub continues!

gouldenstein, 2022-01-24:
I should add, it's not compatible on macOS at least. Perhaps it is on Linux, as per issue #371

mvp, 2022-01-27:
@gouldenstein , before I remove it, can you please post output of uhubctl commands turning off power for ST4300USB3.
Of course, it does not (and it should not) depend on your OS, only on hub hardware.
My main concern is if both USB2+USB3 components are indeed turned off. Be sure to use latest uhubctl. Thanks!

gouldenstein, 2022-01-31:
@mvp hi, here's the code:

```
scottgould@Scotts-iMac ~ % uhubctl -p all -a 0
Current status for hub 20-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [046d:0ab7 Generic Blue Microphones 201701110001]
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [07ca:1311 AVerMedia Technologies, Inc. Live Gamer MINI 0000000000513]
  Port 4: 0100 power
Sent power off request
New status for hub 20-1 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Current status for hub 20-7 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 20-7 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports, ppps]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off

```

commit 53fd1f39 (2022-02-02): Remove StarTech ST4300USB3 from compatible list, does not seem to properly support VBUS off

Closes #390.
closed 2022-02-02 by mvp