mvp/uhubctl #391: Can this tool be used to programatically remove and reinsert USB stick?

takeda, 2022-01-26:
I have a broken USB stick that I try to recover data from. Whenever a bad sector is read the drive appears to fall into an infinite loop (it's internal LED starts rapidly blinking and the stick no longer responds to any commands until it is physically removed and reinserted.

I tried the `-a off` and `-a on` commands and looks like they work, that is, the internal LED turns off and on.

Unfortunately, when the USB stick gets into the failed state, these commands no longer seem to make any effect.

I also noticed that when power is disabled (in normal scenario), the device is not removed by the kernel.

Are there specific option combination that could allow me to do the equivalent of removing and reinserting the stick programmatically?

I'm using FreeBSD 13.0

bigrck64, 2022-01-27:
Maybe you can try ?
`sudo udevadm trigger --action=remove /sys/bus/usb/devices/${location}.${port}/`

See "USB devices are not removed after port power down on Linux" there
https://github.com/mvp/uhubctl

mvp, 2022-01-27:
@takeda , it is Linux kernel bug that upon powering off device may not be removed from /dev, please read this FAQ: https://github.com/mvp/uhubctl#usb-devices-are-not-removed-after-port-power-down-on-linux.
Note that README also lists https://github.com/mvp/uhubctl#notable-projects-using-uhubctl, which has a link to [USB drive data recovery](https://goo.gl/4MddLr) article. That article describes exactly what you need to do to recover your data.

takeda, 2022-01-29:
@mvp @bigrck64 I actually using it on FreeBSD.

To clarify this issue. This is a flash drive not a HD/SSD and looks like a crappy one at that. So I suspect when it encounters a read error it simply locks up (given that LED on it blinks intensively, I'm guessing it is getting into an infinite loop trying to read the block over and over again and never gives up.)

When the lockup happens the only thing that can cure it is physically removing it and putting it back.

When I mentioned that the drive is not removed from the system I should have clarified I meant that it doesn't fully emulate powering off.

I can force removal by calling FreeBSD kernel tool:
```sh
usbconfig -d 1.4 detach_kernel_driver
```

I also forgot to mention another important piece. That while turning off/on appears to work on the USB stick's LED, but when it is in that lock down state (triggered by reading error) despite system sets power to off and it is reporting that it is off, the USB stick continues to blink its LED. This makes believe that disabling/enabling power likely is asking the USB device to turn off its power consumption.

Unfortunately when the stick locks up, it does not respond to any, it's not even responding what device it is anymore.

The data recovery guide you posted, looks like it is very similar case to mine, except in author's situation the disk was slowing down, but was still communicating.

I was hoping that the programmatic way would completely turn power off and back on resetting the device, but that doesn't seem to be the case, or maybe just the controller on my motherboard simply doesn't do that.

mvp, 2022-01-29:
@takeda, it seems that your USB hub doesn't properly support VBUS off, this is why your USB stick doesn't fully reset as you expect it to. I recommend trying to find any hub from supported list, then hook it up to your USB host, insert USB stick to it and run your data recovery process.

I'm keeping process of adding hubs to supported list relatively strict, so only hubs which actually turn off bus voltage completely should be listed.

takeda, 2022-01-29:
Ok, so my suspicion was right. Sorry I wasn't clear at first.

Thank you for your help, I'll close the ticket now.

closed 2022-01-29 by takeda