mvp/uhubctl #393: Add compatible hub (Centech ReTRYHUB - CT-USB4HUB)

netsgnut, 2022-01-28:
Hi,

Just want to report a hub that seems to be working in `uhubctl`. I tried `2.4.0-24-g61187d75`.

Links: [Maker's official web page](http://www.centech.jp/?p=625) (in Japanese), [Amazon.co.jp listing page](https://www.amazon.co.jp/dp/B07234TMNW)

When plugged, `lsusb` reported these devices:

```
Bus 002 Device 010: ID 0424:5744 Standard Microsystems Corp. 
Bus 001 Device 016: ID 0424:2740 Standard Microsystems Corp. 
Bus 001 Device 015: ID 0424:2744 Standard Microsystems Corp. 
```

`uhubctl` output:

```
Current status for hub 2-2 [0424:5744 Microchip Tech USB5744 CEH20060307, USB 3.10, 4 ports, ppps]
  Port 1: 06a0 power Rx.Detect
  Port 2: 06a0 power Rx.Detect
  Port 3: 06a0 power Rx.Detect
  Port 4: 06a0 power Rx.Detect
Current status for hub 1-2 [0424:2744 Microchip Tech USB2744 CEH20060307, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0503 power highspeed enable connect [0424:2740 Microchip Tech Hub Controller]
```

`uhubctl -a off -p 1`:

```
Current status for hub 2-2 [0424:5744 Microchip Tech USB5744 CEH20060307, USB 3.10, 4 ports, ppps]
  Port 1: 06a0 power Rx.Detect
Sent power off request
New status for hub 2-2 [0424:5744 Microchip Tech USB5744 CEH20060307, USB 3.10, 4 ports, ppps]
  Port 1: 04a0 off
Current status for hub 1-2 [0424:2744 Microchip Tech USB2744 CEH20060307, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
Sent power off request
New status for hub 1-2 [0424:2744 Microchip Tech USB2744 CEH20060307, USB 2.10, 5 ports, ppps]
  Port 1: 0000 off
```

`uhubctl -a on -p 1`:

```
Current status for hub 2-2 [0424:5744 Microchip Tech USB5744 CEH20060307, USB 3.10, 4 ports, ppps]
  Port 1: 04a0 off
Sent power on request
New status for hub 2-2 [0424:5744 Microchip Tech USB5744 CEH20060307, USB 3.10, 4 ports, ppps]
  Port 1: 06a0 power Rx.Detect
Current status for hub 1-2 [0424:2744 Microchip Tech USB2744 CEH20060307, USB 2.10, 5 ports, ppps]
  Port 1: 0000 off
Sent power on request
New status for hub 1-2 [0424:2744 Microchip Tech USB2744 CEH20060307, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
```

Do you have anything that you would like us to try and report as well? Thanks!

mvp, 2022-01-28:
Thanks for your report! Can you please check if VBUS off is working? That is, plugged phone stops charging, or USB light stops shining?

netsgnut, 2022-01-28:
Yup, the light stops shining, and whatever device is connected disconnects. That applies to all 4 ports.

mvp, 2022-01-28:
Awesome thanks! I will add this device to supported list

netsgnut, 2022-01-28:
Good to know! :+1: 

Thanks!

commit 950fce23 (2022-02-02): Add Centech CT-USB4HUB as compatible device

Closes #393.
closed 2022-02-02 by mvp