mvp/uhubctl #394: Raspberry Pi Zero 2 W Support

crbn60, 2022-02-01:
I have a Raspberry Pi Zero 2 W and would like to check if it is fully supported. What command output would be help in understanding how it's OTG port is configured?

mvp, 2022-02-01:
Just follow FAQ. But afaik RPI Zero will not work - it was reported multiple times before. You can use any listed hub and connect it to RPi Zero to control power.

mvp, 2022-02-02:
https://github.com/mvp/uhubctl#how-do-i-check-if-my-usb-hub-is-supported-by-uhubctl

closed 2022-02-02 by mvp

crbn60, 2022-02-02:
Just adding some details for the future searcher:

Output of the command on a Raspberry Pi Zero 2 W:

```console
$ sudo uhubctl
No compatible smart hubs detected!
Run with -h to get usage info.
```