mvp/uhubctl #396: Partial Compatibility: BrovSS 10-port USB 3.0 Hub

JulianOrteil, 2022-02-08:
Hi @mvp 

I have a BrovSS 10-port USB hub that has *partial* compatibility with ``uhubctl``. What I mean by this is the hub does not power off of a port but is able to cut the USB signal with the port (all 10 ports appear to be supported). Then after your default 2-second delay (or whatever I specify), the port reconnects. I don't know if total power-off functionality can be implemented on your end somehow, or if we need to reach out to the mfg, but I thought this would be a good addition to your list regardless. :smiley: 

Not specifying any port cuts the signal to all of the devices on the hub, but not the power.

Here is the link to purchase this hub: https://brovss.shop/collections/all-products/products/icone-usb-3-0-hub-10-port-powered-usb-hub-high-speed-data-transfer-aluminum-usb-hub-with-60w-12v-5a-power-adapter-leds-and-mounting-brackets-black

Jules

mvp, 2022-02-09:
Thanks for reporting. However, I'm trying really hard to add only devices which are actually cutting off voltage (VBUS). As of late, many new USB 3 hubs do advertise per port power switching, but tend to save few cents on adding actual power relays to cut off VBUS.
So, I will not add your hub to the list.

I'm really interested to check any new USB-C only hubs. According to USB 3.2 spec, any hub that provides USB-C ports must support true power port switching (it was optional for USB-A). Unfortunately, USB-C only hubs are still extremely rare.

closed 2022-02-09 by mvp

JulianOrteil, 2022-02-09:
Ahhh.. the good ol' "cutting corners for pennies" trick. Nice.

No worries.