mvp/uhubctl #399: Are any of these USB-c hubs?

mgroth0, 2022-02-18:
I haven't checked them all but I checked a bunch and so far the are all usb-b hubs. By USB-c hub, I mean that it is usb-c on both sides. I need a controlable hub for my macbook pro which is connected to monitors via usb c. I'm not sure if usb-b with an adapter would work since it needs to transmit both power and video signals.

mvp, 2022-02-18:
Can you please list which hubs? Link to manufacturer site or amazon page.
Also, please share results for whatever you checked already - preferrably output of running uhubctl, and whether it was able to turn power off.

mgroth0, 2022-02-19:
Sorry I think my original post wasn't clear. I have not ever usde uhubctl and don't own any hubs. All I have done was look through the table on the README, because I want to purchae a usb-c hub that is controllable. But I have been clicking through the list and they all seem to be usb-b, so I wanted to just ask if anyone happens to know if any on the table are usb-c hubs.

mvp, 2022-02-19:
I have not checked any USB-C only hubs, and no one reported them yet.
Most hubs in supported devices table was reported by people checking uhubctl on their hubs.

That said, if you happen to find any USB-C only hub, uhubctl should work because per USB 3.2 spec any hub that provides downstream USB-C ports must support per-port power switching on them - that was not a requirement for USB-A ports before.

Googling for USB-C only hubs gives very few matches, perhaps you could try this one https://www.caldigit.com/thunderbolt-4-element-hub

Also, note that USB-C is quite complicated beast. E.g. ports on macbook are not really USB-C, they are Thunderbolt ports which happen to be able to work in USB mode too.
However, that is not always the case. E.g. some monitors connect to your mac via USB-C connector, but there is no USB protocol used for that connection - they use USB-C connector to run DisplayPort signals to provide a picture to your monitor. And, you cannot connect such monitor to your mac via USB3 hub - it will not work, as such hub cannot route video signal.

That said, latest Thunderbolt4/USB4 hubs like Caldigit should be able to support this variety in full - controlling USB, routing DisplayPort, etc. all via hub itself. And they should fully support per-port power switching too. Please try it and report here. Thanks!

mvp, 2022-02-28:
Closing for inactivity. Please reopen if you have anything to add.

closed 2022-02-28 by mvp

mcuee, 2022-04-05:
Just FYI. I have a Ugreen USB-C to 4 port USB-A ports and one downstream USB-C port. It does not support uhubctl.

dmesg:
```
[ 1878.948414] ucsi_acpi USBC000:00: UCSI_GET_PDOS returned 0 bytes
[ 1879.199650] usb 3-2: new high-speed USB device number 11 using xhci_hcd
[ 1879.350969] usb 3-2: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=92.26
[ 1879.350977] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1879.350980] usb 3-2: Product: USB2.0 Hub
[ 1879.350983] usb 3-2: Manufacturer: GenesysLogic
[ 1879.352186] hub 3-2:1.0: USB hub found
[ 1879.352520] hub 3-2:1.0: 4 ports detected
[ 1879.363876] usb 2-1: new SuperSpeed USB device number 8 using xhci_hcd
[ 1879.386258] usb 2-1: New USB device found, idVendor=05e3, idProduct=0612, bcdDevice=92.26
[ 1879.386266] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 1879.386269] usb 2-1: Product: USB3.0 Hub
[ 1879.386272] usb 2-1: Manufacturer: GenesysLogic
[ 1879.387135] hub 2-1:1.0: USB hub found
[ 1879.387411] hub 2-1:1.0: 4 ports detected

~/build/uhubctl$ sudo ./uhubctl 
No compatible devices detected!
Run with -h to get usage info.
```

lsusb info.
```
lsusb -vvv -d 05e3:0612

Bus 002 Device 008: ID 05e3:0612 Genesys Logic, Inc. Hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0612 Hub
  bcdDevice           92.26
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              1 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0

```

mcuee, 2022-04-05:
Then I have another Ugreen multi-finction USB-C hub with gigabit Ethernet, HDMI, VGA, card reader and 4 USB-A port as well as one type-c port. It does support uhubctl.

```
[ 2255.517017] ucsi_acpi USBC000:00: UCSI_GET_PDOS returned 0 bytes
[ 2256.402627] usb 3-2: new high-speed USB device number 12 using xhci_hcd
[ 2256.502947] usb 2-1: new SuperSpeed USB device number 9 using xhci_hcd
[ 2256.525004] usb 2-1: New USB device found, idVendor=2109, idProduct=0817, bcdDevice= 3.d3
[ 2256.525024] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2256.525027] usb 2-1: Product: USB3.0 Hub
[ 2256.525030] usb 2-1: Manufacturer: VIA Labs, Inc.
[ 2256.526963] hub 2-1:1.0: USB hub found
[ 2256.527141] hub 2-1:1.0: 4 ports detected
[ 2256.552407] usb 3-2: New USB device found, idVendor=2109, idProduct=2817, bcdDevice= 3.d3
[ 2256.552415] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2256.552418] usb 3-2: Product: USB2.0 Hub
[ 2256.552420] usb 3-2: Manufacturer: VIA Labs, Inc.
[ 2256.553396] hub 3-2:1.0: USB hub found
[ 2256.553566] hub 3-2:1.0: 4 ports detected
[ 2257.082873] usb 2-1.1: new SuperSpeed USB device number 10 using xhci_hcd
[ 2257.108345] usb 2-1.1: New USB device found, idVendor=05e3, idProduct=0749, bcdDevice=15.35
[ 2257.108353] usb 2-1.1: New USB device strings: Mfr=3, Product=4, SerialNumber=5
[ 2257.108356] usb 2-1.1: Product: USB3.0 Card Reader
[ 2257.108358] usb 2-1.1: Manufacturer: Generic
[ 2257.108360] usb 2-1.1: SerialNumber: 000000001536
[ 2257.114047] usb-storage 2-1.1:1.0: USB Mass Storage device detected
[ 2257.114377] scsi host0: usb-storage 2-1.1:1.0
[ 2257.199201] usb 2-1.3: new SuperSpeed USB device number 11 using xhci_hcd
[ 2257.236129] usb 2-1.3: New USB device found, idVendor=0bda, idProduct=0412, bcdDevice= 1.43
[ 2257.236138] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2257.236142] usb 2-1.3: Product: 4-Port USB 3.0 Hub
[ 2257.236144] usb 2-1.3: Manufacturer: Generic
[ 2257.245736] hub 2-1.3:1.0: USB hub found
[ 2257.247722] hub 2-1.3:1.0: 2 ports detected
[ 2257.294478] usb 3-2.3: new high-speed USB device number 13 using xhci_hcd
```

uhubctl
```
sudo ./uhubctl 
Current status for hub 3-2.3 [0bda:5412 Generic 4-Port USB 2.0 Hub, USB 2.10, 2 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
Current status for hub 3-2 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [0bda:5412 Generic 4-Port USB 2.0 Hub, USB 2.10, 2 ports, ppps]
  Port 4: 0100 power
Current status for hub 2-1.3 [0bda:0412 Generic 4-Port USB 3.0 Hub, USB 3.10, 2 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 0203 power 5gbps U0 enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001536]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [0bda:0412 Generic 4-Port USB 3.0 Hub, USB 3.10, 2 ports, ppps]
  Port 4: 0203 power 5gbps U0 enable connect [0bda:8153 Realtek USB 10/100/1000 LAN 001000001]

```
```

mvp, 2022-04-05:
Ugh. I guess manufacturers are not keen on actually implementing USB specs 👎