mvp/uhubctl #400: Framework Laptop?

Animeshz, 2022-02-19:
It shows: `No compatible devices detected!` even when running as the root user.

The device pid:vid is `1d6b:0003` and looks like its on the supported devices list in the README.md.

```sh
  ~ >>> lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 27c6:609c Goodix Technology Co., Ltd. Goodix USB2.0 MISC
Bus 003 Device 005: ID 046d:c52b Logitech, Inc. Unifying Receiver
Bus 003 Device 004: ID 8087:0032 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

mvp, 2022-02-19:
Many OEMs use the same USB chipsets. Each chipset can be configured to enable or disable per port power switching. Apparently this laptop manufacturer decided not to - nothing we can do.

closed 2022-02-19 by mvp

Animeshz, 2022-02-20:
@mvp No, I mean I was able to disable all ports from the BIOS, probably its not exposed to the kernel or something else, can you please tell a little bit of info how this tool works? I want to help to integrate this tool in framework too, their prime mission align with these stuffs, they've open sourced their EC firmware too.

mvp, 2022-02-20:
This tool is querying USB hubs for if they support per port power switching as defined in USB 2.0 and later specifications. If USB hub claims to support it, it will try sending appropriate USB control messages to the hub to switch port power off or on. Unfortunately, this laptop doesn't advertise USB per port power switching support, that's why it is not detected by uhubctl.

davidscaife, 2024-10-11:
I don't mean to revive old threads but thought it might be useful for @Animeshz and/or others looking for this to note that the ports on the Framework do actually seem to support power switching with this tool despite not being automatically detected (as stated above, `lsusb -v` unfortunately reports "No power switching" for each).

But, if you use the `--force` flag, uhubctl seems to work fine in cutting power to the individual ports. I don't know how consistent it is between builds but for me, the ports are 7-1 (rear left), 5-1 (rear right), 1-2 (front left) and 1-1 (front right), each toggling on+off happily on my Framework 13 (AMD).

mvp, 2024-10-12:
If hub doesn't report power switching, powering off with --force will disconnect USB devices, but it is quite unlikely that it will actually disable VBUS (5v USB voltage). You can test it by voltmeter or checking if USB can stops spinning.

davidscaife, 2024-10-12:
Ah, indeed. My bad, I had tested with a device that was only charging, but was also being "clever" about it, not charging unless it had the data connection. Testing again with a device that only uses the 5v power lines shows it is still providing power as you said. Thanks for the clarification