mvp/uhubctl #404: Cypress CY4608 HX2VL

virtualhere, 2022-03-04:
This hub seems to only support PPPS on port 1. The other ports advertise PPPS but only port 1 actually powers down.

mvp, 2022-03-07:
Maybe this is the reason? 
```
Bus 001 Device 060: ID 04b4:6570 Cypress Semiconductor Corp. Unprogrammed CY7C65632/34 hub HX2VL
```
note `Unprogrammed`.
This hub is development kit, and maybe programmed differently by developers?

virtualhere, 2022-03-07:
No, unprogrammed just means that the vendorid/productid etc has not been set to a vendors real id. i.e the "blaster" program used to write to the eeprom has not been run. By default is is set to ppps however but this is unfortunately not fully implemented in hardware except for port 1

mvp, 2022-03-07:
Perhaps. But still, development kits are designed for OEMs to create finished products and to be highly configurable. They are not intended to be used by wide public like us.

commit 7496d53f (2022-03-07): Add a note for Cypress CY4608 HX2VL

Closes #404.