mvp/uhubctl #405: Raspberry Pi 4B USB ports not ganged???

emilla67, 2022-03-04:
Hello.

Maybe I am dreaming ... 

I have a Pi 4B, rasbian buster 32 bit OS, updated to 04/03/2022.

The only tweak that I did in the OS is adding "usbcore.autosuspend=-1" by the end of /boot/cmdline.txt (and reboot). I think it has nothing to do with PPPS ..., but I am not an expert.

Ouput of sudo uhubctl is:
Current status for hub 2 [1d6b:0003 Linux 5.10.63-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, **ppps**]
  Port 1: 0080 off
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, **ppps**]
  Port 1: 0000 off
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 5.10.63-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, **ppps**]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]

Notice "ppps" + port 1 is off and the rest are on.

I tested to connect a mobile in port 1 and it is not charging. Same mobile in port 2, 3 or 4 is charging.

Did anyone else observe this behaviour?

BRs,

Esteban.

emilla67, 2022-03-07:
Hi mvp !

First of all, thanks for uhubctl development !!!

Detailing a bit more about the environment if it helps:
Raspberry Pi 4B, r 1.4 (as stated when I bought it), 8 GB (I have read that this 8GB Pi has something different to other lower RAM Pi's, but I didn't understand it fully and I am not able to find it again + also something related to VL805 firmware upgrade, which I also don't understand).
  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

I don't have a voltmeter or USB light, but I tested the same mobile in the past (last year, 2021,  for sure) and yes I agree that it was all 4 ports on or off. But now, what I am experiencing is different: all 4 ports can be turned on or off individually and the mobile tells me if it is charging or not ... (the battery icon in the mobile that says if is charging or not, and it is coherent to the port on/off uhubctl status !!!).

I can provide the output of any commands suggested + testing with the limited capabilities that I have.

Again, thanks a lot !!!

BRs,

Esteban.

mvp, 2022-03-07:
unless some miracle happened and raspberry pi has separate power switches to each port, it is most likely this:
> It is possible that phone will stop charging by absence of data lines, even if voltage was provided.

emilla67, 2022-03-08:
Hi mvp !

Yeah, you are right. I managed to do the test with a portable battery pack that has blinking lights when it is charging, and yes power is on even if the port has been powered off with uhubctl.

Therefore, it is the mobile that stops charging by absence of data lines, as you suggested.

On the other hand, for my use case (I have 2 mobiles connected to the Pi ports 1 and 2 and I want to manually and individual control if charge one mobile, the other, both or any), it is enough.

Thanks !!!

Esteban.

closed 2022-03-13 by emilla67