mvp/uhubctl #406: Add another compatible usb hub

Mr-Bossman, 2022-03-06:
P/N: juh377
product site: https://en.j5create.com/products/juh377
first port has its own power control(p4), then (2,3) are independent controlled by p2.
On the second 4 port controller (4,5) are controled by p1, and (6,7) are p3.

The 7th port has a sticker in-front which hides 2 screws to open the housing. On the board it seems to have over current protection and the pins for each ports power control are broken out to resistors.

The GL3523 QFN76 is the usb hub controller

mvp, 2022-03-07:
Thanks for reporting, added to the list.

closed 2022-03-07 by mvp

Tbruno25, 2022-10-21:
This hub only removes the power for a brief time and does not maintain a non-powered state.

mvp, 2022-10-21:
@Tbruno25 , are you sure you are not hitting Linux kernel issue as [described in FAQ](https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux)?

Tbruno25, 2022-10-21:
Great point. Although other hubs seem to function as expected I cannot be certain -- will revisit the faq to confirm and report back.

Mr-Bossman, 2022-10-21:
I can take a look in a couple of days if nessary.

Tbruno25, 2022-10-21:
@mvp my fault -- should have paid closer attention to the faq! supplying the `-r 1000` args seems to have fixed it for me :facepalm: 

Interesting that this behavior only occurs with certain hubs...

mvp, 2022-10-21:
Thanks. This behavior should be linked not to the hub, but to device being connected - because Linux attaches different device drivers per each device type, and they may behave differently when they want to revive dead device.

Note that this issue is solved in Linux kernel 6.0 - uhubctl built from master has support for new Linux power off interface now.

teamdoug, 2023-04-19:
Has anyone tested this when acting as a USB3 hub? Cutting power works when the hub is plugged into a USB2 port but not USB3 for me.

Mr-Bossman, 2023-04-19:
Will look into it. If I recall correctly I did test both

Mr-Bossman, 2023-04-21:
> Has anyone tested this when acting as a USB3 hub? Cutting power works when the hub is plugged into a USB2 port but not USB3 for me.
This happens for me too. Not sure why it doesn't work on usb3 anymore. very odd that if you plug it in using a usb2 cable it works. @mvp any ideas.
I tried with -S and -e.

commit 6f1c71d5 (2022-03-07): Add j5create JUH377 as compatible device

Closes #406.