mvp/uhubctl #407: RSHTech/Rosonway 7 port

proppy, 2022-03-08:
https://www.rshtech.com/products/7-port-aluminum-usb-c-hub-with-individual-on-off-switchesrsh

# dmesg
```
[118936.108143] usb 4-1: new SuperSpeed USB device number 28 using xhci_hcd
[118936.156088] usb 4-1: New USB device found, idVendor=2109, idProduct=0817, bcdDevice=90.14
[118936.156096] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[118936.156100] usb 4-1: Product: USB3.0 Hub             
[118936.156103] usb 4-1: Manufacturer: VIA Labs, Inc.         
[118936.156105] usb 4-1: SerialNumber: 000000000
[118936.157714] hub 4-1:1.0: USB hub found
[118936.157810] hub 4-1:1.0: 4 ports detected
[118936.280081] usb 2-5: new high-speed USB device number 48 using xhci_hcd
[118936.429751] usb 2-5: New USB device found, idVendor=2109, idProduct=2817, bcdDevice=90.14
[118936.429760] usb 2-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[118936.429764] usb 2-5: Product: USB2.0 Hub             
[118936.429767] usb 2-5: Manufacturer: VIA Labs, Inc.         
[118936.429770] usb 2-5: SerialNumber: 000000000
[118936.430261] hub 2-5:1.0: USB hub found
[118936.430375] hub 2-5:1.0: 4 ports detected
[118937.048263] usb 4-1.3: new SuperSpeed USB device number 29 using xhci_hcd
[118937.096147] usb 4-1.3: New USB device found, idVendor=2109, idProduct=0817, bcdDevice=90.14
[118937.096155] usb 4-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[118937.096159] usb 4-1.3: Product: USB3.0 Hub             
[118937.096162] usb 4-1.3: Manufacturer: VIA Labs, Inc.         
[118937.096165] usb 4-1.3: SerialNumber: 000000000
[118937.108347] hub 4-1.3:1.0: USB hub found
[118937.109261] hub 4-1.3:1.0: 4 ports detected
[118937.468077] usb 2-5.3: new high-speed USB device number 49 using xhci_hcd
[118937.617879] usb 2-5.3: New USB device found, idVendor=2109, idProduct=2817, bcdDevice=90.14
[118937.617887] usb 2-5.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[118937.617891] usb 2-5.3: Product: USB2.0 Hub             
[118937.617894] usb 2-5.3: Manufacturer: VIA Labs, Inc.         
[118937.617897] usb 2-5.3: SerialNumber: 000000000
[118937.618407] hub 2-5.3:1.0: USB hub found
[118937.618521] hub 2-5.3:1.0: 4 ports detected
```

# uhubctl
```
👹 sudo ./uhubctl -l 2-5 -l 4-1
Current status for hub 4-1 [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [2109:0817 VIA Labs, Inc. USB3.0 Hub 000000000, USB 3.20, 4 ports, ppps]
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-5 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
```

# lsusb -v

```
Bus 004 Device 029: ID 2109:0817 VIA Labs, Inc. USB3.0 Hub             
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x0817 
  bcdDevice           90.14
  iManufacturer           1 VIA Labs, Inc.         
  iProduct                2 USB3.0 Hub             
[... 163 lines cut ...]
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat           4 micro seconds
    bU2DevExitLat         231 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {30eef35c-07d5-2549-b001-802d79434c30}
Device Status:     0x000d
  Self Powered
  U1 Enabled
  U2 Enabled
```

Switch port on and off seemed to work, but I did hit https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux

mvp, 2022-03-29:
Sorry for late response, I missed your submission :-(
Can you please verify if VBUS off is working properly for this hub? That is, connected phone stops charging or USB fan stops spinning when you turn port power off?

closed 2022-05-11 by mvp

theodesu, 2022-05-21:
Hi! I'm testing the RSH-518 (non-C), and while reporting correctly uhubctl, the VBUS does not cut to any port, even when setting the whole hub to "off". Tested on a raspberry pi 4B, both in USB 3 and USB 2 ports on the pi, and with and without external power. I think the only difference in the C-version is that a USB C cable is included, instead of USB A, which makes me sceptical that this hub should work...

Below is how the hub reports:
`sudo uhubctl
Current status for hub 2 [1d6b:0003 Linux 5.10.60-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.3.3 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.3 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 5.10.60-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]`

`lsusb -v

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.10
  iManufacturer           3 
  iProduct                2 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
[... 219 lines cut ...]
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12`

paul-walmsley-sifive, 2024-05-01:
There appear to be multiple versions of this hub with different chipsets.  Perhaps some support software-controlled per-port power switching and others don't.  I purchased an RSH-518 here that has a Realtek chipset in it, and uhubctl is unable to control port power.

olalonde, 2026-10-20:
Confirming the plain **Rosonway RSH-518** (no "C") does **not** cut VBUS. This is the Realtek-based variant.

**Hardware**
- Hub: Rosonway RSH-518, 7 ports, USB 3.0, per-port power buttons, externally powered
- Chips: two cascaded Realtek RTS5411, USB IDs `0bda:5411` (USB 2) and `0bda:0411` (USB 3)
- Host: Raspberry Pi 4 Model B Rev 1.5, kernel `6.18.50+rpt-rpi-v8`
- uhubctl 2.6.0 (Debian package `2.6.0-1`)

**Topology** (the hub is 2 chained 4-port chips; `1-1.2` port 1 feeds `1-1.2.1`):
```
1-1.2    0bda:5411  USB2.1 Hub   (USB3 dual: 2-2   0bda:0411)
1-1.2.1  0bda:5411  USB2.1 Hub   (USB3 dual: 2-2.1 0bda:0411)
```

**Test:** a USB lamp on one of the external ports.
```
sudo uhubctl -l 1-1.2   -p 2,3,4   -a off
sudo uhubctl -l 1-1.2.1 -p 1,2,3,4 -a off
```
uhubctl detects both chips as `ppps` and reports every port `off` on both the USB 2 and USB 3 halves. **The lamp stays on.** Retrying with `-r 20` changes nothing.

The reverse fails too: after switching the lamp's port off with the hub's own button, `-a on` doesn't bring it back. Port power seems to follow only the physical buttons, and the RTS5411's port-power control isn't wired to anything.

So for anyone buying: the **RSH-518C** (VIA VL817, `2109:0817`) is the variant reported working above; the plain RSH-518 with Realtek chips is not. Might be worth a note in the compatibility table.

commit a0f8c87e (2022-05-11): Add Rosonway RSH-518C as compatible device

Closes #407.