mvp/uhubctl #408: Raspberry Pi 4B USB Ports Repower Automatically

kenjfox, 2022-03-11:
uhubctl is able to shut off usb power, but only for a few seconds.  Something is causing the devices to be "found" and repowered.  Is there something I need to do to disable the auto-discovery of the devices?

Mar 10 17:25:02 vangoedge mtp-probe[2476]: checking bus 1, device 7: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar 10 17:25:02 vangoedge kernel: logitech-hidpp-device 0003:046D:404D.000B: input,hidraw0: USB HID v1.11 Keyboard [Logitech K400 Plus] on usb-0000:01:00.0-1.3/input2:1
Mar 10 17:25:02 vangoedge kernel: input: Logitech K400 Plus as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.2/0003:046D:C52B.000A/0003:046D:404D.000B/input/>
Mar 10 17:25:02 vangoedge mtp-probe[2474]: bus: 1, device: 7 was not an MTP device
Mar 10 17:25:02 vangoedge mtp-probe[2474]: checking bus 1, device 7: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Mar 10 17:25:02 vangoedge kernel: usb 1-1.1: USB disconnect, device number 6
Mar 10 17:25:02 vangoedge kernel: logitech-djreceiver 0003:046D:C52B.000A: hiddev96,hidraw1: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:01:00.0-1.3/input2
Mar 10 17:25:02 vangoedge kernel: usb 1-1.3: Manufacturer: Logitech
Mar 10 17:25:02 vangoedge kernel: usb 1-1.3: Product: USB Receiver
Mar 10 17:25:02 vangoedge kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Mar 10 17:25:02 vangoedge kernel: usb 1-1.3: New USB device found, idVendor=046d, idProduct=c52b, bcdDevice=24.07
Mar 10 17:25:02 vangoedge kernel: usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
Mar 10 17:25:01 vangoedge kernel: usb 1-1.3: USB disconnect, device number 4
Mar 10 17:24:59 vangoedge sudo[2456]: pam_unix(sudo:session): session closed for user root
Mar 10 17:24:59 vangoedge sudo[2456]: pam_unix(sudo:session): session opened for user root(uid=0) by pi(uid=0)
**Mar 10 17:24:59 vangoedge sudo[2456]:     root : TTY=pts/0 ; PWD=/root ; USER=root ; COMMAND=/usr/sbin/uhubctl -l 1-1 -a 0**

Debian GNU/Linux 11 (bullseye)
uhubctl 2.4.0
BOOTLOADER: up to date
   CURRENT: Tue 25 Jan 2022 02:30:41 PM UTC (1643121041)
    LATEST: Tue 25 Jan 2022 02:30:41 PM UTC (1643121041)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1

mvp, 2022-03-11:
Yes, this is FAQ https://github.com/mvp/uhubctl#power-comes-back-on-after-few-seconds-on-linux

closed 2022-03-11 by mvp

kenjfox, 2022-03-11:
Thanks!  and  sorry!