mvp/uhubctl #411: RSHTech/Rosonway RSH-A16

Dampen59, 2022-04-11:
Hello,
First thing, thank you for the work you put into uhubctl !

I have a problem with RSH-A16 (listed as compatible hub). I only can power-switch the first 6 ports.
All the other 10 ports can't be powered-off, only data lines are cut.
Tested using smartphone charge + USB fan

Here's the picture of the packaging and the hub itself :

[image]
[image]
[image]

The hub is connected to a Raspberry Pi 4 USB 2.0 port using the original HUB cable
OS : Raspberry PI OS

Here are both `lsusb` output without and with the HUB plugged in the USB 2.0 port using the original HUB cable

Without :
```
pi@raspberrypi:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

With :
```
pi@raspberrypi:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 084: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 083: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 082: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 081: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 080: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Output of `sudo uhubctl` :
```
pi@raspberrypi:~ $ sudo uhubctl
Current status for hub 2 [1d6b:0003 Linux 5.15.32-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.4.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.4.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1 [1d6b:0002 Linux 5.15.32-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
```

Thank you

mvp, 2022-04-11:
1. What version of uhubctl you are using? Some old raspberry pi distro carry very old uhubctl. You need 2.4.0
2. Raspberry Pi supports USB 3, why not use USB 3 all the way?

Dampen59, 2022-04-11:
Thank you for your reply.

I use the latest version available, as I cloned the repo on the Raspberry Pi and used "make"
```
pi@raspberrypi:~ $ sudo uhubctl --version
2.4.0-30-g6f1c71d5
```

I just did an other test on a computer with Ubuntu 20.04 installed.
I used the same USB hub and cloned uhubctl repo.
The HUB is connected through an USB 3 port.

I have the same result.
Some ports works nicely, others don't.

Dampen59, 2022-04-11:
Physical positions start at 1 from RSTECH logo (cables side)
Physical Positions 1 to 6 are OKAY
<google-sheets-html-origin><style type="text/css"></style>

Physical Loc | Location | Port Param
-- | -- | --
1 | 1-1.4 | 2
2 | 1-1.4 | 1
3 | 1-1.4.3 | 2
4 | 1-1.4.3 | 1
5 | 1-1.4.3 | 3
6 | 1-1.4.3 | 4
7 | 1-1.4.4 | 2
8 | 1-1.4.4 | 1
9 | 1-1.4.4.3 | 2
10 | 1-1.4.4.3 | 1
11 | 1-1.4.4.3 | 3
12 | 1-1.4.4.3 | 4
13 | 1-1.4.4.4 | 2
14 | 1-1.4.4.4 | 1
15 | 1-1.4.4.4 | 3
16 | 1-1.4.4.4 | 4

Dampen59, 2022-04-12:
Update : using Ubuntu 20.04, latest uhubctl
USB hub connected to USB 3 port
Test material : an USB powered desk fan

Tested on physical port 1
Location 1-3 
Port 2
Command used `sudo uhubctl -a on -l 1-3 -p 2`
Also tried with `-r` parameter

Test not passed.

`lsusb` output
```
Bus 002 Device 011: ID 2109:0822 VIA Labs, Inc. 
Bus 002 Device 010: ID 2109:0822 VIA Labs, Inc. 
Bus 002 Device 009: ID 2109:0822 VIA Labs, Inc. 
Bus 002 Device 008: ID 2109:0822 VIA Labs, Inc. 
Bus 002 Device 007: ID 2109:0822 VIA Labs, Inc. USB3.1 Hub             
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 111: ID 046d:c092 Logitech, Inc. G203 LIGHTSYNC Gaming Mouse
Bus 001 Device 097: ID 2109:2822 VIA Labs, Inc. G203 LIGHTSYNC Gaming Mouse
Bus 001 Device 096: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 095: ID 2109:2822 VIA Labs, Inc. G203 LIGHTSYNC Gaming Mouse
Bus 001 Device 094: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 093: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub             
Bus 001 Device 002: ID 17ef:608c Lenovo Lenovo Calliope USB Keyboard
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`sudo uhubctl` output
```
Current status for hub 2-1.4.4 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1.4.3 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1.4 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 4: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
Current status for hub 2-1.3 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-1 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 4: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
Current status for hub 1-3.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-3.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-3.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-3.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
```

mvp, 2022-04-13:
Sorry it takes long time to test this on my side - I have recently moved to new place, and my RSH-A16 is sitting somewhere in moving boxes still unpacked. Once I get my hands on it, I will test it and if VBUS behavior on first 6 ports is confirmed, I will update README. Thanks for your report!

Dampen59, 2022-04-14:
Hello, thank you for your reply.

Don't be sorry, take your time.

I emailed the RSHTECH support, there's an interesting sentence in their reply :

> We understand that you want to control the port via uhubctl. In the past two years, due to the impact of covid-19, the supply chain of chipsets has become extremely challenging. We adopted extra chip solutions to ensure a more stable product supply at the retail end.

I ordered the first HUB on February, 2022 the second this month (April, 2022)
Maybe I do have a different hardware revision on both HUBs due to their shortage of chipsets, but I don't have anything to confirm that. It seems that there are no screws to open it to have a look.

I will receive today a 10p Amazon USB HUB (the one listed in uhubctl README), maybe I will have better luck with this one.

mvp, 2022-05-11:
Finally I found my RSH-A16 and it works for me 100%.

My hub was sent directly to me by RSHTech and is looking quite differently from yours.
First, it has different USB vid:pid, namely: `0bda:0411` (Realtek), yours showing `2109:0822` (Via).
USB chipset replacement alone should be big red flag.
Second, your physical port assignment is also different. Mine is (a-b is local path prefix and may differ across different hosts):

| Physical port | Location | Port param | PPPS |
|:-----------:|:---------:|:---------:|:-----:|
| 1 | a-b | 1 | OK |
| 2 | a-b | 2 | OK |
| 3 | a-b.4 | 1 | OK |
| 4 | a-b.4 | 2 | OK |
| 5 | a-b.4 | 3 | OK |
| 6 | a-b.4 | 4 | OK |
| 7 | a-b.3 | 1 | OK |
| 8 | a-b.3 | 2 | OK |
| 9 | a-b.3.4 | 1 | OK |
| 10 | a-b.3.4 | 2 | OK |
| 11 | a-b.3.4 | 3 | OK |
| 12 | a-b.3.4 | 4 | OK |
| 13 | a-b.3.3 | 1 | OK |
| 14 | a-b.3.3 | 2 | OK |
| 15 | a-b.3.3 | 3 | OK |
| 16 | a-b.3.3 | 4 | OK |

I have also performed test for each and every port by connecting a phone and issuing uhubctl power off then on.
For each of 16 ports, power was actually cut, and blue LED went off.

I assume it means RSHTech is now producing different device batches?
I recommend reaching out to their support.
I will link this issue in README for reference.

closed 2022-05-11 by mvp

mvp, 2022-05-11:
Note that my hub has Rosonway on top, and yours has RSHTech:

[image]

[image]