mvp/uhubctl #413: Switch on RP4b not working (worked on raspberry b+ first gen)

nightyx, 2022-04-24:
Hey there,
I have an unusual issue and did not find any solutions online. 

I used uhubctl on a raspberry b+ first gen successfully with an Rosonway RSH-A16. Admittedly I didnt test all ports, but during development I could switch the ports on the 'furthest' hubs/ports (e.g. 1-1.4.4.4).

Then I upgraded to a raspberry 4b ("Raspberry Pi 4 Model B Rev 1.5") and now the switching does not work. 
- I have the latest version of uhubctl (I cloned it from the github page and installed with make)
- I have VL805 firmware version 000138a1

I have the feeling that the power-switching is not send to the RSH-A16 hub. Example: I have two USB devices connected (one is a single usb lightbulb, and the other one is the RSH-A16 (containing more lightbulbs). When I turn off all ports (e.g. via `uhubctl -l 1-1 -a 0`), then the lightbulb turns of (as expected), however the other lightbulbs in the RSH-A16 hub are still turned on.

I tried using the repeated turning-off thingy (e.g. `uhubctl -l 1-1 -a 0 -r 1000`) and that did not work.
I tried forcing the power-off with `-f` and that did not work as well.
I tried using the usb hub on usb2 and usb3 ports. makes no difference.

Checking `lsusb -v` it shows, that the RSH-A16 supports power switching (at least, that the raspberry correctly identifies that the usb hub is capable of it).

`Bus 002 Device 006: ID 2109:0822 VIA Labs, Inc. USB3.1 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x0822
  bcdDevice           90.14
  iManufacturer           1 VIA Labs, Inc.
  iProduct                2 USB3.1 Hub
  iSerial                 3 000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes           19
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Feedback
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               8
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
  bPwrOn2PwrGood      175 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  bHubDecLat          0.4 micro seconds
  wHubDelay          2292 nano seconds
  DeviceRemovable    0x00
 Hub Port Status:
   Port 1: 0000.02a0 lowspeed L1
   Port 2: 0000.02a0 lowspeed L1
   Port 3: 0000.02a0 lowspeed L1
   Port 4: 0000.02a0 lowspeed L1
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x002a
  bNumDeviceCaps          3
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      HIRD Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat           4 micro seconds
    bU2DevExitLat         231 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {30eef35c-07d5-2549-b001-802d79434c30}
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x000d
  Self Powered
  U1 Enabled
  U2 Enabled`

I´m not sure what my mistake might be. As I said, the same usb hub worked on the raspberry b+ first gen before but now doesnt work on the 4b. TBH it worked once briefly and then after a reboot it did not work again. 
Any ideas on what to try to make it work?

mvp, 2022-04-24:
What version of uhubctl you have? Anything older than 2.4.0 is not likely to work on RPi4.

mvp, 2022-04-24:
Ah sorry I missed that you are using latest version of uhubctl - just make uhubctl -v shows 2.4.0-something.

As for reasons why it doesn't work, one possiblity is that usb2/usb3 post matching code may not able to do find proper port pairing. You can bypass it with -e switch, but then you have to send turn off commands to appropriate usb2 and usb3 ports twice. Can you show uhubctl output when you turn off port which doesn't turn off? It should be displaying commands sent to usb2, then usb3 busses.
If newer RPi4 changed USB topology, it may theoretically break.
Actually, can you just paste output of running uhubctl without any parameters?

nightyx, 2022-05-02:
here is the output without any parameters

```
pi@raspberrypi:~ $ uhubctl
Current status for hub 2-2.4.4 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2.4.3 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2.4 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 4: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
Current status for hub 2-2.3 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 2-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 4: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
Current status for hub 2 [1d6b:0003 Linux 5.15.32-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 0263 power 5gbps U3 enable connect [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.2.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.2.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1.2.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.2 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 5.15.32-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431
```

mvp, 2022-05-11:
It seems that you have wrong batch of RSH-A16. I have tested mine and it defintely works great.
However, some people are now reporting different kind of RSH-A16 sold, and it does not seem to work.
More in issue #411. I will close this as duplicate I guess.

closed 2022-05-11 by mvp

nightyx, 2022-07-23:
Hey there,
i finally got the chance to test the RSH-A16 on the old raspy again and it works! Thus I´ll reopen this one. So now I would think, as the usbhub works fine, that either the script or the raspbi4b is not working properly. 
It would be nice to make it work on the raspi4b, as my old b+ is acting up ;) So any idea on how to solve it for the 4b? Anything you would need from me?

Here are some outputs:

Its a very old raspberry b+ first gen
`Raspberry Pi Model B Plus Rev 1.2`
`Linux raspberrypi 5.10.63+ #1488 Thu Nov 18 16:14:04 GMT 2021 armv6l`

```
pi@raspberrypi:~ $ sudo uhubctl -v
2.4.0-24-g61187d75
```

output of `lsusb -v` interestingly doesn't show anything related to powerswitching, but the output itself looks way different than the one on the raspy4b, so I guess it's due to the very old linux version / version of the lsusb package ;) 
```
Bus 001 Device 007: ID 2109:2822 VIA Labs, Inc. USB2.0 Hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x2822
  bcdDevice           90.14
  iManufacturer           1 VIA Labs, Inc.
  iProduct                2 USB2.0 Hub
  iSerial                 3 000000001
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Single TT
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      2 TT per port
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
```

```
pi@raspberrypi:~ $ sudo uhubctl
Current status for hub 1-1.4.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.4.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1 [0424:9514, USB 2.00, 5 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0424:ec00]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 5: 0100 power
Current status for hub 1 [1d6b:0002 Linux 5.10.63+ dwc_otg_hcd DWC OTG Controller 20980000.usb, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0424:9514, USB 2.00, 5 ports, ppps]
```

power switching works as expected
```
pi@raspberrypi:~ $ sudo uhubctl -l 1-1.4.4.4 -p 2 -a off
Current status for hub 1-1.4.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
Sent power off request
New status for hub 1-1.4.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 2: 0000 off
pi@raspberrypi:~ $ sudo uhubctl -l 1-1.4.4.4 -p 2 -a on
Current status for hub 1-1.4.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 2: 0000 off
Sent power on request
New status for hub 1-1.4.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
```

I just did an apt-get upgrade/update on the 4b, but still power switching is not working on my raspi with the RSH-A16. Happy to do testing if you have any ideas on how to make it work.

nightyx, 2022-07-23:
as an additionall comment.. looking at this discussion: https://github.com/mvp/uhubctl/issues/417 my rsh-a16 has the same (faulty?) chips. However: power switching on the old raspi b+ works flawlessly. Switching with a raspi4b is not working.

mvp, 2022-07-23:
I don't quite get what the issue is @nightyx. RSH-A16 works when connected to old RPiB, but not when connected to RPi4B? In that case I would recommend trying to use USB2 only cable to force hub USB2 connection to RPi4B

nightyx, 2022-07-23:
thanks for your quick reply

> RSH-A16 works when connected to old RPiB, but not when connected to RPi4B?
yes, thats the current issue. And as the old b+ is acting up, I´d like to switch to the 4b :) 

Great idea using a usb2 cable. I tried that, and the output is this:
```
pi@raspberrypi:~ $ sudo uhubctl
Current status for hub 2 [1d6b:0003 Linux 5.15.32-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.1.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.1.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.1.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1.1.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 5.15.32-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0507 power highspeed suspend enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
```

However switching did not seem to work. The output says, that the ports were switched off, but that didnt affect the usb-lights unfortunately. e.g.:
```
pi@raspberrypi:~ $ uhubctl -l 1-1.1.4.3 -a off
Current status for hub 1-1.1.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Sent power off request
New status for hub 1-1.1.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
```

Do i need to turn off the internal usb3 hub stuff on the raspi4b for this? I didnt do that yet, since this should affect the external usbhub, right?

mvp, 2022-07-23:
Another idea is to disable xhci. It seems that old hardware would use ehci or uhci, and new would always use xhci (even for USB2 forced connection). I don't remember how to disable xhci exactly, must be either via kernel bootup options or modules.conf setting

TexHexx, 2022-07-26:
Hi, do you have Rosonway or RSHTech on the top of the hub? Cause I am having RPi4b also and wanted to order this hub with Rosonway brand on the top.

mvp, 2022-07-26:
@TexHexx, I have Rosonway branded RSH-A16 with USB vendor id `0bda`, it seems to always work for everything.

nightyx, 2022-07-26:
@TexHexx I have the RSHTech-branded version

TexHexx, 2022-08-16:
Hi @nightyx, i just got my hub. It is really not working with RPi4b. Mine is branded as Rosonway, but chips are the same "VIA Labs". What RP did you use before on witch it was working?

TexHexx, 2022-08-16:
@mvp i think they change chips to VIA everywhere, data is cutting but power is not on RPi4

nightyx, 2022-08-16:
> Hi @nightyx, i just got my hub. It is really not working with RPi4b. Mine is branded as Rosonway, but chips are the same "VIA Labs". What RP did you use before on witch it was working?

I have the Raspberry b+, first generation. 
Unfortunately I don't have access to other raspberries, so I can't test more.

TexHexx, 2022-08-16:
@nightyx no way, i took even RPi 3b+ now to test and also not working ...

TexHexx, 2022-08-17:
@nightyx @mvp hi guys, tested with RPi3b+ again deeply, working ok, cutting data and power fully on all ports. Will retest now all on RPi4b+ again. Device detected as 2109:2822 Via. On the top of it is Rosonway brend.

nightyx, 2022-08-17:
@TexHexx sounds great. What did you do to make it work on the RPi3b+?

TexHexx, 2022-08-17:
@nightyx nothing special, reinstall OS again, did update & upgrade and install uhubctl from sources

TexHexx, 2022-09-07:
Hello @mvp, after lot of tests i found out that stable work is for ports with has from 1 to 3 digits, ports with 4 digits not working. So from 1 to 8 ports power switching ok in usb 2.0 mode, in usb 3.0 no port switching at all, but ports from 9 to 16 - can not switch power. Is it possible to do something with it? Can attach any logs if needed.

mvp, 2022-09-07:
Sorry, not much can be done from uhubctl side if your hardware is misbehaving.
There is very small possibility that uhubctl gets confused with USB2/3 duality and tries to switch off wrong hub pair, but that is only applicable to USB3 mode.
If you are unable to control ports in USB2 only mode, that must be hub fault.
Since this is USB3 hub, there is a possшbility that its firmware can be updated.
However, I do not see Rosonway offering firmware updates. Some companies like Plugable do offer firmware updates for their USB3 hubs, not sure if those can be used instead - most likely not.