mvp/uhubctl #415: How to operate devices

Michdo93, 2022-05-11:
Hi,

I have probably a sau dämmliche question. How do I use it to run my devices? On a Raspberry Pi is clear to me, I install the program and turn off and on the USB ports. But how do I now control any USB hub? This is a very general question. Do I just plug it into my computer via USB cable? With a Raspberry Pi, for example, I have 600mA. With a USB hub with 10 ports, this would be very weak. I have also configured this on the Raspberry Pi to 1.2A. With 4 devices, I still get over the 1.2A limit and nothing works. Since I can exclude a Raspberry Pi as an end device rather. Then I tried my notebook. There I then come to 1.5A. That is enough for maybe a single Smarthpone that I then charge.

Let me describe this differently. If I connect a USB hub to the socket, then I rarely have a problem to operate multiple devices, charge or whatever. However, I lack a microcontroller with which I could control and turn off individual ports. If I have to connect this USB hub to a computer first, then I am limited by its USB port. I can turn off individual ports, but the amperes might be too high for what the computer could supply.

Mhm... I think the scenario still to end. If I connect a USB hub via USB, I can control this. Are there USB hubs that draw power from the outlet and then I get access to from a computer so I could turn off individual ports?

I'm thinking something like Programmable USB Hubs or Managed Smart USB Hubs. Or maybe a USB hub that can be accessed via RS232 or another interface. But there would be a different method again. With RS232 for example, I would not have to use their program but serial commands that the manufacturer has specified.

Considering how many devices really use their ports as a hub, so that I can only switch them all on and off together, or how many devices are limited by the maximum amperage, I come to the conclusion that this only works cleanly and without problems for very few individual devices. Or do I understand the principle wrong?

If we take the AmazonBasics devices as an example, they all just draw power from my PC, which has a limit to what it can put out.

Thanks in advance.

mvp, 2022-05-11:
Yes, you just connect supported hub to your computer. Note that your suported hub is almost always required to have its own power supply (sometimes it's quite considerable, e.g. RSH-A16 has 100W power supply). Very few devices are able to work without own power supply (e.g. Apple keyboard), but those typically can only support 1-2 light devices like mouse.

AmazonBasics hubs have their own power supply too.

mvp, 2022-05-11:
> Considering how many devices really use their ports as a hub, so that I can only switch them all on and off together, or how many devices are limited by the maximum amperage, I come to the [...]

No. Almost all supported hubs must provide their own power to each downstream USB port, e.g. 0.9A for USB3 hubs.
And they should support power control for each port independently. It means for say 7 port USB3 hub they should have 0.9A * 5V * 7 = 31W power supply to be able to power all 7 devices. And 16 port hub must have at least 0.9A * 5V * 16 = 72W, but e.g. RSH-A16 actually provides 100W, slightly above the requirement.

closed 2022-05-11 by mvp

Michdo93, 2022-05-12:
Thank you, when I saw your reply in the e-mail, I was still slightly puzzled. The further explanation was helpful. So the RSH-A16 has a USB port for my computer and an extra power port so that the USB ports get enough power.

I thought I would have to go with something more expensive like Acroname's managed USB hubs. But then you get something usable for 1/10 of the price.