mvp/uhubctl #416: Time to perform enumeration

p-sousa, 2022-05-17:
Is there some way to run the uhubctl multiple times without the need to perform enumeration each time?
I would like to use the tool but in each run enumeration (finding hubs) can take > 5 seconds and the actual toggling of the USB ports is very fast, just wondering if there is some way to make it faster assuming the USB devices connected do not change.

mvp, 2022-05-17:
uhubctl relies on libusb API to perform it's actions, and unfortunately any
libusb action requires descriptor that can be obtained only by using full
devices enumeration.

That said, you still have few options:

* uhubctl supports multiple ports operation e.g. -p a,b,c. That allows you
to speed it up 4x+.
* You can call uhubctl in parallel threads of your app with different
target ports, it will make it work faster overall
* Some OS maintain internal cache of connected USB devices, e.g. MacOS
definitely does

mvp, 2022-05-18:
Check this for python example https://github.com/mvp/uhubctl/discussions/355

p-sousa, 2022-05-18:
@mvp thanks a lot for the answer!
I am using it in Ubuntu 20.04 and it doesn't seem to cache it at all. I am considering creating a daemon, it would enumerate in the beginning ( and ideally monitors for changes.. if something is plugged/unplugged) and then we can request operations through a unix socket or something. Do you see any problem or a better way? Would you be interested in a Pull Request with this functionality?

mvp, 2022-05-19:
libusb has limited hotplug support on some architectures, so device watch can be implemented in uhubctl, but will be quite a bit of effort. But before you jump to implementing anything, I have to ask your final objective, what are you trying to achieve? What is exactly the purpose of your daemon? To keep devices turned off as they are detected? Or something else? What is not sufficient for you in current uhubctl usage, except for enumeration performance? Maybe we can solve this performance bottleneck differently?

Also, uhubctl is designed to be as simple and straightforward command line as possible, and is currently working on all known platforms (Windows is unfortunate exception due to Microsoft not enabling kernel  support for hub port power control). Making uhubctl into daemon will significantly deviate from this goal of being simple tool.

That said, I don't mind accepting your PRs after we talk through the objectives.

p-sousa, 2022-05-19:
My only issue is the enumeration performance, it takes a lot of time. About the daemon, I get your point it's a lot of complexity to add almost nothing to the tool.
Do you think it would be possible to cache it somehow after the first run? For example, dumping the `hubs` array to a file and having an option to provide that file in the next run.

mvp, 2022-05-19:
Unfortunately answer is no, this is libusb limitation: https://libusb.sourceforge.io/api-1.0/group__libusb__dev.html
> libusb has a concept of a USB device, represented by the libusb_device opaque type. A device represents a USB device that is currently or was previously connected to the system. Using a reference to a device, you can determine certain information about the device (e.g. you can read the descriptor data).

> The libusb_get_device_list() function can be used to obtain a list of devices currently connected to the system. This is known as device discovery. Devices can also be discovered with the hotplug mechanism, whereby a callback function registered with libusb_hotplug_register_callback() will be called when a device of interest is connected or disconnected.

So, before uhubctl can do anything with particular USB device (even if it knows full USB location and port number), it has to obtain pointer to libusb_device, which currently can be found only by `libusb_get_device_list()` (can be slow), or by hotplug callback.

In my experience, uhubctl never takes more than 1-2s to change port state. Also, you can call multiple instances of uhubctl in parallel, and they will work fast as long as you target different ports. And, as I mentioned in first response, you can operate all ports at the same hub (location) in one uhubctl invocation, e.g. `uhubctl -l 3-4.5 -p 1,4,5` - it will change ports 1,4 and 5 very quickly.

mvp, 2022-06-18:
Closing for inactivity. Also, this has a working solution described in #355.

closed 2022-06-18 by mvp