mvp/uhubctl #417: RSHTech/Rosonway RSH-A16 sometimes working

theodesu, 2022-05-22:
Hello,

Thanks for your great work with this software, it is very useful. Hopefully I can add some insight to help other users. I've tested the RSH-A16 hub, and like Dampen59 mentioned in [issue 411](https://github.com/mvp/uhubctl/issues/411), I'm seeing different vendor/device ID's [2109:2822].

However, my first 8 ports always work, and the other 8 _sometimes_ work. It reports power off, but the port doesn't cut the VBUS power. It seems to report indentically when it's working and not working. I've tested this on a raspberry pi 4, and a macbook pro 2020 with the same result. Plugging it out and back in again, or via uhubctl cycling the port that the entire hub is connected to, can make it work. My guesstimate is about a 30% chance that it will work. I've tried disabling autosuspend in the os, but that does not make a difference. I'm also only using USB 2.

It seems strange that this is an intermittent issue, as I thought that the support was mainly a hardware implementation issue

Any pointers on where to look would be greatly appriciated. Also, if I somehow can detect if it works or not through software, I could build a service that resets it until it works on boot.

## uhubctl
```
Current status for hub 2 [1d6b:0003 Linux 5.10.60-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-[1.4.4.4] [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-[1.4.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.4.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0503 power highspeed enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0103 power enable connect [1915:521f Nordic Semiconductor Secure DFU Bootloader 14221a740824ca44]
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0503 power highspeed enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1 [1d6b:0002 Linux 5.10.60-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
```

## sudo lsusb -v
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.10
  iManufacturer           3 Linux 5.10.60-v7l+ xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:01:00.0
  bNumConfigurations      1
  Configuration Descriptor:
[... 992 lines cut ...]
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

mvp, 2022-05-22:
Few questions:
1. Extra brackets for hub location - is this real or copy/paste issue?
```
Current status for hub 1-[1.4.4.4] [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
```
2. Your hub has Via chipset, but my fully working hub has Realtek chipset. Do you have it branded as RSHTECH or Rosonway? It seems that Via chipset is faulty for power control. I recommend contacting RSHTECH support to clarify if they sell different device batches, and if it possible to get Realtek device.
3. When you control power, is there a chance you control ports used for daisy chaining internal 4 port hubs? If you did, that would likely to lead to instability. You need to be very careful to never change state for daisy chaining ports.

theodesu, 2022-05-23:
Hi, and thanks for your reply!

1. Yes, copy/paste issue, it looks normal in my terminal.
2. It is branded RSHTECH. I'll get in touch with them, thanks.
3. I only touch the ports I'm intending to control, not the daisy chained hubs. However, I have tried doing it intentionally, in hopes of resetting an unresponsive hub, but it has not had any effect.

mvp, 2022-06-18:
Added a warning link to device table pointing to this issue.

closed 2022-06-18 by mvp

slk9, 2022-09-16:
I have Identically looking RSH-A16 bought within 2 month difference on Amazon from the same seller. One has 0bda:0411 chipset and works as expected. Latest one has 2109:0822 chipset and when recognized by uhubctl and disconnects USB devices still keep power on ports ON :(.

commit 5418edf9 (2022-06-18): Add a warning link that RSHTech RSH-A16 may not work

Rosonway RSH-A16 works, while RSHTech RSH-A16 may not.
Closes #417.