mvp/uhubctl #419: Getting unexpected hub addresses on macOS 12.5 Beta

dyhopper, 2022-05-24:
We have a testing setup where we daisy-chain 2 smart hubs and have sensors plugged into the child hub that we turn on/off with this excellent tool.
This has always given us addresses like 20-3.1.1 (for port 1 on child hub when this hub was plugged into port 1 of parent). However with macOS 12.5 Beta using the same machine and hub setup, we only get hub addresses with 2 levels. For example we get 20-3, 20-4, 0-3, 0-4, 20-3, and 0-1 with our current setup.  Note that 20-3 is listed twice. 

Again, this is macOS BETA, so it's possible it will be "fixed" when macOS comes out of beta. Also, I can likely work around this if needed. But I thought I'd bring it up in case uhubctl might need some updates for macOS 12.5.

mvp, 2022-05-25:
Thanks for heads up!
We had similar breakage for MacOS 10.5 https://github.com/libusb/libusb/issues/707.
This is not `uhubctl` problem, but `libusb` problem - `uhubctl` relies on `libusb_get_port_numbers` working correctly.
Back then it was fixed by changing Mac specific `libusb` code in https://github.com/libusb/libusb/pull/712.
I expect that we will have to do something similar if it is not fixed in beta.

mvp, 2022-06-18:
Lets close this for now. May need to re-open when MacOS 12.5 is released and if this issue is present in it.

MPLew-is, 2022-06-18:
This exact issue is also occurring for me on 12.4, can this be reopened?

mvp, 2022-06-18:
@MPLew-is, can you please give more info about your hardware: Mac model, OS version, libusb version, uhubctl version, how did you build/install both libusb and uhubctl? If you have libusb older than current 1.0.26 and uhubctl older than 2.4.0, please try installing latest versions first from homebrew. Thanks!

MPLew-is, 2022-06-20:
[image]
[image]

Already fully up to date, with both `libusb` and `uhubctl` installed via Homebrew.

mvp, 2022-06-20:
I can also reproduce it on MacOS 12.4 (21F79).
This is some OS change that broke libusb `libusb_get_port_numbers()`.
Easiest way to reproduce is to build libusb and run `examples/listdevs` - it should list device path with dots present.
I will dig a bit more into this and file a bug against libusb.

mvp, 2022-06-20:
Filed this issue with libusb https://github.com/libusb/libusb/issues/1156

mcuee, 2022-06-26:
Strange that for my Mac Mini M1 running macOS 12.4, both libusb listdev and uhubctl seem to work fine.
* https://github.com/libusb/libusb/issues/1156#issuecomment-1166474660

MPLew-is, 2022-07-20:
This seems to be fixed on macOS 12.5 that dropped today:

[image]

mvp, 2022-07-20:
Issue closed, workaround is to upgrade to MacOS 12.5. Noted in README too.

commit 54248b8f (2022-07-20): Note that issue #419 is fixed in MacOS 12.5

Closes #419.
closed 2022-07-20 by mvp