mvp/uhubctl #422: LAN9514: 'OFF' is delayed up to 'ON' command

joelguittet, 2022-06-07:
Hello,

I'm using LAN9514 from Microchip. USB ports have USB-switches as recommend in the application note and as shown on the BeagleBoard-xM (https://github.com/beagleboard/beagleboard-xm/blob/master/BeagleBoard-xM_SCH.pdf).

When powering OFF 1 port, it remains On (lsusb still showing the wifi dongle connected on it). Then when sending the ON command, it seems power is switched OFF and then ON. The USB device disconnects and reconnects in dmesg logs.

Why this strange behavior ? Is somebody able to test this on the BeagleBoard-xM ?

Thanks,
Joel

mvp, 2022-06-08:
This is most likely this FAQ entry https://github.com/mvp/uhubctl#usb-devices-are-not-removed-after-port-power-down-on-linux
Please check that VBUS gets turned off. After you use uhubctl to turn off port, you cannot trust lsusb output, as kernel doesn't propagate knowledge about port being turned off up to userspace.

joelguittet, 2022-06-09:
Hello @mvp 
Thanks for this clarification and your quick answer.
I haven't realized there was the answer in the wiki. I have checked with a multimeter today and this is my exact case.
Closing the issue.
Joel

closed 2022-06-09 by joelguittet