mvp/uhubctl #423: Rosonway RSH-A13 13 ports USB 3.1 2109:2822 works with uhubctl

Kryd0s, 2022-06-17:
Output from uhubctl:
`Current status for hub 2 [1d6b:0003 Linux 5.15.32-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.3.4.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.3.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
Current status for hub 1-1.3.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Current status for hub 1-1.3 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [0bda:2838 Realtek RTL2838UHIDIR 00000001]
  Port 3: 0507 power highspeed suspend enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 4: 0100 power
Current status for hub 1 [1d6b:0002 Linux 5.15.32-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
`

Tested each port. Data and power are cut correctly with --action off:
`Physical Port Location
1        2    1-1.3
2        1    1-1.3
3        2    1-1.3.3
4        1    1-1.3.3
5        3    1-1.3.3
6        4    1-1.3.3
7        2    1-1.3.4
8        1    1-1.3.4
9        2    1-1.3.4.3
10       1    1-1.3.4.3
11       3    1-1.3.4.3
12       4    1-1.3.4.3
13       4    1-1.3.4
`

Tested with Raspberry Pi 4 Model B 4GB. Let me know if you need any other information. Thanks.

mvp, 2022-06-18:
Thank you for reporting, @Kryd0s! Can you please test it while connected via native USB3 connection?
Currently your hub is connected via USB2 only.
uhubctl will show USB3 devices in addition to USB2 ones you have listed now, something like this:
```
Current status for hub 2-2.3.3 [0bda:0411 Generic 4-Port USB 3.1 Hub, USB 3.20, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2.3.3 [0bda:5411 Generic 4-Port USB 2.1 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
...
```

Kryd0s, 2022-06-18:
> Can you please test it while connected via native USB3 connection? Currently your hub is connected via USB2 only.

Yeah, you're right. I edited the original post to have the USB3 data. For anyone who has a RPi 4 Model B the blue USB ports are the USB3 and the black ones are USB2.

Kryd0s, 2022-06-18:
I did some more testing with this USB hub. The above data is from a Raspberry Pi 4 Model B 4 GB plugged into the USB3 port, and all 13 ports can be powered on and off perfectly. Out of curiosity though I hooked the hub up to my older Raspberry Pi 2 Model B 512 MB, and I'm getting some strange results.

It seems like locations with 3 or 4 digits work. So all of the 1-1.3 ports, and 1-1.3.3 ports work, but all four of the 1-1.3.4.3 ports don't work. I send the off command and the output from the terminal all looks correct, but the USB port doesn't actually lose power.

Both the RPi 2 and RPi 4 are running the same latest version of Raspberry Pi OS bullseye 32 bit with all the latest updates done. Any ideas what I can try to do to make the 5 digit locations work on the RPi 2?

mvp, 2022-06-18:
Since you have Via chipset, it does seem like this hub is similar in behavior to RSHTech RSH-A16 as described in issue #417. Rosonway RSH-A16 (which looks identical, but branded Rosonway and has Realtek USB chipset) does work properly on all ports...

Kryd0s, 2022-06-18:
> It seems like locations with 3 or 4 digits work. So all of the 1-1.3 ports, and 1-1.3.3 ports work, but all four of the 1-1.3.4.3 ports don't work. I send the off command and the output from the terminal all looks correct, but the USB port doesn't actually lose power.

So it definitely doesn't have anything to do with 4 digits or 5 digits. I unplugged and plugged the hub into the same port on the RPi over and over and eventually it was able to control all 13 ports. It really does behave exactly as described in issue https://github.com/mvp/uhubctl/issues/417.

I'll do some more testing with it on my RPi 4 and see if it works 100% of the time there.

mvp, 2022-07-02:
Added this hub to the list - since it works most of the time. We may need more reports to really understand what is going on here.

closed 2022-07-02 by mvp

slk9, 2022-10-06:
Works well for me with all 13 ports when A13 HUB comes with Realtek chipset 0bda:0411

Matheus-Garbelini, 2023-01-04:
@slk9 did you get your hub from amazon?

slk9, 2023-01-09:
Yes, I bought it from Amazon as recommended by support. It works well on my Mac. I have seen some issues on Ubuntu 20.04 so when after power off/on device not recognized by host anymore.

CpServiceSpb, 2023-11-05:
Does it mean that when port goes to off (-a off) appropriate port light is off also ?

Kryd0s, 2023-11-05:
@CpServiceSpb Yes, the blue light goes off when the uhubctl command turns the power off to that port.

CpServiceSpb, 2023-11-09:
Can you check out by USB tester are there some ampers and volst on a port ehen uhubctl mives it to iff ?
Is it zero/zero or not ?

CpServiceSpb, 2023-11-29:
I bought RSHTech A13 based on Via822 from Aliexpress: [RSHTech A13](https://aliexpress.ru/item/1005005801695589.html?spm=a2g2w.orderlist.0.0.43004aa6AcV1Cc&sku_id=12000035552202085).
There are 4 Via822 chips.
From cord side: 1st chip, 2nd chip, 3rd chip, 4th chip.
As I suppose, hub interconnection schema is the following: two hub chips are connected to 3 and 4 ports od 1st hub chip, and one hub chip is connected to 4 port of 2nd hub chip.

And I confirm that all 9 ports on hubs connected to 1st hub works, but 4 ports on hub connected to 2nd hub which is conneced to 1st hub doesn' t work.
That is 4 ports with maximum figures id in comparing of other id figures of the device doesn' t work !

I wrote to the seller at Aliexpress regarding thi situation.
The answer was it is hub chip issue.
Maybe such cascading - chip to chip to chip to cord - causes the situation or maybe there are not some elements on these ports.
I cannot imagine.
Who can tell exactly ?

skasianov, 2025-03-17:
Latest A13  (not old RSH-A13-5) with "2109:2822 VIA Labs, Inc." chipset does not allow to control power :(

commit 38ff0320 (2022-07-02): Add Rosonway RSH-A13 as compatible device (with some caveats)

Closes #423.