mvp/uhubctl #425: Works but also doesn't?!

git-avinash, 2022-06-23:
I am trying to switch on and off my Huawei 4g dongle. My hub has a power indicator next to it, it shuts down during the cycle process and comes back up. But the light on my dongle keeps blinking meaning it still didn't switch off completely! I want to completely power off the dongle and turn it back on. How can I do it?

Specs
`
Raspberry pi 4B 8GB
Debian GNU/Linux 11 (bullseye)
`

uhubctl
`
Current status for hub 2 [1d6b:0003 Linux 5.15.32-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.4 [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [12d1:14dc HUAWEI_MOBILE HUAWEI_MOBILE]
  Port 2: 0503 power highspeed enable connect [12d1:14dc HUAWEI_MOBILE HUAWEI_MOBILE]
  Port 3: 0503 power highspeed enable connect [12d1:14dc HUAWEI_MOBILE HUAWEI_MOBILE]
  Port 4: 0100 power
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2813 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
Current status for hub 1 [1d6b:0002 Linux 5.15.32-v8+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
`

The command I am using to cycle
`
location=1-1.4; port=1; mode=2; delay=10; echo 0 > sudo tee /sys/bus/usb/devices/${location}.${port}/authorized; uhubctl -a ${mode} -l ${location} -p ${port} -d ${delay} -r 200
`

dmesg
`
[46253.727017] cdc_ether 1-1.4.1:1.0 eth2: unregister 'cdc_ether' usb-0000:01:00.0-1.4.1, CDC Ethernet Device
[46254.458498] usb 1-1.4.1: USB disconnect, device number 111
[46267.059624] usb 1-1.4.1: new high-speed USB device number 112 using xhci_hcd
[46267.173379] usb 1-1.4.1: New USB device found, idVendor=12d1, idProduct=14dc, bcdDevice= 1.02
[46267.173409] usb 1-1.4.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[46267.173426] usb 1-1.4.1: Product: HUAWEI_MOBILE
[46267.173441] usb 1-1.4.1: Manufacturer: HUAWEI_MOBILE
[46267.185693] cdc_ether 1-1.4.1:1.0 eth2: register 'cdc_ether' at usb-0000:01:00.0-1.4.1, CDC Ethernet Device, 0c:5b:8f:27:9a:64
[46267.187053] usb-storage 1-1.4.1:1.2: USB Mass Storage device detected
[46267.187944] scsi host1: usb-storage 1-1.4.1:1.2
[46268.192400] scsi 1:0:0:0: Direct-Access     HUAWEI   TF CARD Storage  2.31 PQ: 0 ANSI: 2
[46268.192795] sd 1:0:0:0: Attached scsi generic sg1 type 0
[46268.193227] sd 1:0:0:0: Power-on or device reset occurred
[46268.193726] sd 1:0:0:0: [sdb] Media removed, stopped polling
[46268.194938] sd 1:0:0:0: [sdb] Attached SCSI removable disk
`

mvp, 2022-06-23:
This is FAQ, for RPi4B you have to turn off all 4 ports https://github.com/mvp/uhubctl#raspberry-pi-4b
```
uhubctl -l 2 -a 0
```
or cycle with
```
uhubctl -l 2 -a 2
```
You do not need to change authorized state for cycling. Fiddling with authorized state may be helpful if you want to keep it from coming back on its own - which is actually not on its own, but kernel USB driver may want to re-power it, which for power cycle does not matter.

closed 2022-06-23 by mvp

mvp, 2022-06-23:
Per FAQ: https://github.com/mvp/uhubctl#raspberry-pi-turns-power-off-on-all-ports-not-just-the-one-i-specified

### Raspberry Pi turns power off on all ports, not just the one I specified
This is the limitation of Raspberry Pi hardware design. As a workaround, you can buy any external USB hub from supported list above, attach it to any USB port of Raspberry Pi, and control power on its ports independently. Also, there are supported hubs designed specifically for Raspberry Pi, e.g. UUGear MEGA4.

If LED is blinking, `-r 1000` might help, or it might be possible that even when off RPi4B still provides trickle current that is enough to keep your modem on. That said, RPi4B is definitely supporting VBUS off, so it should work if you use `sudo uhubctl -l 2 -a 0`.

git-avinash, 2022-06-23:
I'm aware of the limitation, that's why I'm already using an externally powered usb hub. Everything is working like I mentioned. It does turn of the port I specify. I know it because port led turns off on the hub. But for some reason dongle still receiving current and won't turn off properly.

git-avinash, 2022-06-23:
I'm using this hub:
https://www.amazon.in/gp/product/B011OSM2KW/

git-avinash, 2022-06-23:
I used this command again
```
location=1-1.4; port=1; mode=0; echo 0 > sudo tee /sys/bus/usb/devices/${location}.${port}/authorized; uhubctl -a ${mode} -l ${location} -p ${port} -r 1000
```

Check out this video to know what's actually happening:
https://youtube.com/shorts/VBcCaxNDp44?feature=share

As you can see, the port-led turned off but dongle still blinking

mvp, 2022-06-23:
I was under impression that you plug your dongles into RPi4B directly - that should work, as RPI4B does support VBUS off if you turn off all 4 ports.

Note that your QZ Active Powered USB 3.1 Hub is not in supported list - it most likely does not even support VBUS off properly - if it did, voltage would be completely removed once turned off by uhubctl.

I recommend to get any hub from supported list instead. Or, at least quick test is to plug it into RPi4 directly and see if you can kill power completely.

git-avinash, 2022-06-24:
damn, indeed! it works properly when I connect a dongle to pi itself and turn off all ports

SlySven, 2022-06-26:
It might be worth pointing out that `uhubctl` causes the specified hub device to (try and) remove power from its output(s) but that does not have any effect on downstream devices if they are self-powered rather than drawing power from that device - and since they will stop receiving USB data they cannot then be turned off if any of the devices upstream of them are deactivated!