mvp/uhubctl #429: Port can be powered off, but device is still enumerated (also Microchip USX2064 hub is working)

jeremyherbert, 2022-07-10:
Hi,

I am using a device: https://www.seeedstudio.com/reterminal-with-cm4-p-4904.html

with a built in Microchip [USX2064](https://www.microchip.com/en-us/product/USX2064) USB2 hub:

```
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         2 TT per port
  bMaxPacketSize0        64
  idVendor           0x0424 Standard Microsystems Corp.
  idProduct          0x2514 USB 2.0 Hub
  bcdDevice            b.b3
  iManufacturer           0
  iProduct                0
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0029
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      1 Single TT
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       1
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      2 TT per port
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      1 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0100 power
   Port 2: 0000.0100 power
   Port 3: 0000.0100 power
   Port 4: 0000.0100 power
Device Qualifier (for other device speed):
  bLength                10
  bDescriptorType         6
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         0 Full speed (or root) hub
  bMaxPacketSize0        64
  bNumConfigurations      1
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered
```

`uhubctl` does indeed work to remove power from the individual ports (only 2 are actually exposed on the device) - the power LED on the two devices I tested turns off.

The interesting thing though is that when powering off the device, the powered off device still shows up in `lsusb`, even if I unplug the USB cable, and (in the case of my test device) the USB serial port is still enumerated as ttyACM0. There are no messages or anything in `dmesg`. If I unplug the cable to the device, and then power the port on again with `uhubctl`, then finally the "USB disconnected" message shows up in `dmesg`. Is this normal behaviour?

mvp, 2022-07-10:
Thanks for your report. This is Linux USB kernel driver issue: https://github.com/mvp/uhubctl#usb-devices-are-not-removed-after-port-power-down-on-linux
I will add this hub to the list.

mvp, 2022-07-12:
Added Seeed Studio reTerminal CM4104032 to compatible list.
We do not add built-in hubs into the list as they often differ between end user devices.
Instead, we add only end-user devices as something that can be bought on open market.

commit 06e7240d (2022-07-12): Add Seeed Studio reTerminal CM4104032 as compatible device

Closes #429.
closed 2022-07-12 by mvp