mvp/uhubctl #430: Getting Log floods when turning USB-Power off

Stathogon, 2022-07-14:
Hello, the Error that floods my Logs are 
`kernel: [22646073581] usb 1-1: clear tt 1 (9041) error -71`
this will quickly fill up my 80 mb log partition (5 mins).
Im running that on a 4 GB Pi 4 b and using this Command to controll the Power:
`sudo uhubctl -l 2 -a 0`
Any recommandation on what is going wrong?

Stathogon, 2022-07-14:
Using eeprom version: 000138a1

mvp, 2022-07-14:
I have also seen weird things happening when using bad USB cables - try using different cable.

mvp, 2022-07-20:
Closing by timeout after one week. Please reopen if you have anything to add.
Seems like this is your local issue - no one else reported it before, please provide more details if you can reliably reproduce it on more than one RPi4 sample.

closed 2022-07-20 by mvp