mvp/uhubctl #431: RPi 4b switching

daxliniere, 2022-07-14:
Hello Mikhail,
Thanks for this project.

I have an RPi 4b and wish to be able to power-cycle my USB peripherals. I've gone through the tests you specify here and got the result below from **lsusb -v**. The thing is, I wouild be happy with ganged switching, since it's not imperative that the other 3 USB ports stay powered up.

What is the command for power-cycling a ganged power switching hub, please?

Thank you in advance.

All the best,
Dax.

==========================
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         3
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.10
  iManufacturer           3 Linux 5.10.103-v7l+ xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 0000:01:00.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
        bMaxBurst               0
Hub Descriptor:
  bLength              12
  bDescriptorType      42
  nNbrPorts             4
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
[... 229 lines cut ...]
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x02
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0001
  Self Powered

mvp, 2022-07-14:
RPi4B is already supported - you need to turn off all 4 ports for power state to really change
```
sudo uhubctl -l 2 -a 0
```
If you have another external USB hub which only supports ganged switching, you can force controlling it with `-f` (`--force`) option - be sure to control all ports for it to work (but this is still long shot afaik)

closed 2022-07-14 by mvp

daxliniere, 2022-07-14:
Thank you, Mikhail, much appreciated!!

daxliniere, 2022-07-15:
Hi @mvp, when I run **uhubctl -l 1-1 -a 2** on my RPI-4B, I get the error "No compatible smart hubs detected at location 1-1!"
According to this, I don't think that should happen, right?

I'm running latest firmware (138?).

mvp, 2022-07-15:
@daxliniere, please provide the output of following commands:
* `sudo uhubctl -v`
* `sudo uhubctl -f`

daxliniere, 2022-07-15:
> * `sudo uhubctl -v` - 2.0.0
> * `sudo uhubctl -f` - invalid option

mvp, 2022-07-15:
uhubctl supports RPi4B only since version 2.4.0. anything older will not work

daxliniere, 2022-07-16:
Okay, that's strange to me. I only installed uhubctl a day before I started posting here. I ran **sudo apt install uhubctl**

daxliniere, 2022-07-16:
Any reason the package that RPi has access to is out-of-date? Is that something you can fix, or does it need to be fixed by RP Foundation?

mvp, 2022-07-16:
Sorry, but I don't control how often my software is updated by various Linux distros. You need to reach out to your distro provider. Or, you can compile latest version from source per FAQ.

daxliniere, 2022-07-16:
Okay, now running 2.4.0-40-g06e7240d

Now I get this for **sudo uhubctl -f**

```
Current status for hub 2 [1d6b:0003 Linux 5.10.103-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.1 [050d:0234, USB 2.00, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [088e:5036 iLok iLok 6E8CA02A_492C0003]
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [050d:0234, USB 2.00, 4 ports, ppps]
  Port 2: 0100 power
  Port 3: 0303 power lowspeed enable connect [096e:0201 OEM USB DONGLE 27D58FB09832C8ED]
  Port 4: 0103 power enable connect [064f:03e9 WIBU-SYSTEMS AG CodeMeter-Stick 0001044222]
Current status for hub 1 [1d6b:0002 Linux 5.10.103-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 2.00, 1 ports, ppps]
  Port 1: 0503 power highspeed enable connect [2109:3431 USB2.0 Hub, USB 2.10, 4 ports, ppps]

```

mvp, 2022-07-16:
It looks like you have supported 4-port USB2 connected to your RPi4  - `[050d:0234, USB 2.00, 4 ports, ppps]`. You should be able to control its ports using `sudo uhubctl -l 1-1.1 -p {n} -a 0`.
If you want to control ports on RPi4 built-in hub, use `sudo uhubctl -l 2 -a 0`.

mvp, 2022-07-16:
If you want to avoid using sudo all the time, there is a section in FAQ that explains how to fix USB permissions