mvp/uhubctl #434: RSHTECH RSH-A104 working on USB2

sjmtlewy, 2022-07-26:
Just tested an RSHTECH RSH-A104 and it works perfectly, but only when not connected to USB3. e.g. to a USB2 port or via a USB1/2 only cable. 

Hub was bought this week from Amazon UK.

Ports can be switched individually and VBUS turns off.

mvp, 2022-07-26:
Which version of uhubctl you tried and on what hardware?
Please make sure you have uhubctl 2.4.0, and if using RPi4B, update firmware before making any conclusions.
uhubctl should work on USB3 just as well as USB2.
If it does not, please provide output of following commands:
```
uhubctl --version
sudo uhubctl
```
And also output of your `uhubctl` commands which you used to turn power off/on.
When you state does not work, please clarify what exactly does not behave properly.
Thanks!

sjmtlewy, 2022-07-27:
uhubctl 2.4.0-35-ga6159d68
I turn off the ports with ./uhubctl -a 0 -l ... 
Ubuntu 18.04 running on an x86 Gigabyte Brix mini PC

When turning off the ports and connected over USB3, the tool reports that the ports have been switched off (USB2 and corresponding USB3 ports), but VBUS remains on.

My test device is a USB3 charge doctor. The hub is USB3, but my PC has both USB2-only and USB3 ports. I have also tried connecting the hub to the USB3 port but with a USB2-only cable, which works (indicating this may be an issue relating to the use of USB3 rather than problem with specific ports).

sjmtlewy, 2022-07-27:
Sorry, I've not got the hub on me now, but I will be getting hold of another one for more trials soon

sjmtlewy, 2022-07-27:
Curiously, I also tried an RSH-A10, which would appear identical to the A13 and A16 models that have been reported working, but it did not support per port switching at all. The problem we hit is that the model number of a hub is just branding and the internals are liable to be changed at whim

mvp, 2022-07-27:
Please provide output for `sudo uhubctl` with this hub connected. Thanks!

sjmtlewy, 2022-08-05:
output:
`Current status for hub 2-4 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
`

sjmtlewy, 2022-08-05:
USB2 and USB3 ports report as being "off" when setting either to off, but VBUS power is not cut. When the hub is only connected to USB2, it works perfectly.

I have not experienced the inconsistent behaviour reported by other users of RSHTech hubs - it works very reliably on USB2 (which is good enough for me) and never on USB3

sjmtlewy, 2022-08-05:
lsusb -v output:

[lsusb.txt](https://github.com/mvp/uhubctl/files/9267605/lsusb.txt)

mvp, 2022-08-05:
Thanks @sjmtlewy , added this hub to the table!

commit e1e4d450 (2022-08-05): Add Rosonway RSH-A13 as compatible device

This works only if connected via USB2.

Closes #434.
closed 2022-08-05 by mvp