mvp/uhubctl #437: Generic AliExpress hub works

johngleeson, 2022-08-18:
Hi, 

I have the 7 port version of [this](https://www.aliexpress.com/item/1005002558956981.html) hub, and I can toggle on/off all ports. It looks to be a generic hub that's listed under many brands. Interestingly, there's some funky mapping. It's split into two logical banks, with one nested inside the other. Starting numbering at the downlink side:

Bank 1 
1,2,3
4 is ganged for all of bank 2

`uhubctl -a off -p 0 -f -l 2-1` will turn off the rightmost port

Bank 2
1,2,3,4

`uhubctl -a off -p 4 -f -l 2-1.4` will turn off the leftmost port

mvp, 2022-08-22:
This hub looks too cheap to actually support VBUS off. Can you please test that it supports it? Easiest way is to turn power off with uhubctl and check if connected phone stops charging

mvp, 2022-08-31:
Closing by inactivity from submitter. Please feel free to reopen if you can test VBUS off support

closed 2022-08-31 by mvp

johngleeson, 2022-09-01:
Sorry for the delay. You're correct. VBUS off support does not work. I tested using my phone, as you described.