mvp/uhubctl #440: D-Link DUB-H4 rev. B1 report PPPS, but does not switch power on any port

thedolphin, 2022-08-22:
P/N: DUBH4E....B1
HW Ver: B1

```
Bus 002 Device 070: ID 05e3:0660 Genesys Logic, Inc. USB 2.0 Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x05e3 Genesys Logic, Inc.
  idProduct          0x0660 USB 2.0 Hub
  bcdDevice            5.07
  iManufacturer           0
  iProduct                1 USB2.0 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00e9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 32 FS bits
    Port indicators
  bPwrOn2PwrGood       50 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0000
   Port 2: 0000.0000
   Port 3: 0000.0000
   Port 4: 0000.0000
```

mvp, 2022-08-22:
DUBH4E sounds weird. We have listed DUB-H4, not H4E.
Can you please post a picture of your hub, preferrably with model and hw rev on it. Thanks!

thedolphin, 2022-08-23:
[image]
[image]

mvp, 2022-08-23:
This is so weird. I do remember checking DUB-H4 rev A silver back in 2008, and it didn't support VBUS off. I also was getting reports that DUB-H4 rev B is new black model. However, you obviously have DUB-H4 rev B silver, which should not even exist 😁.
Not sure what to do here. Perhaps we should remove DUB-H4 from supported table completely...

thedolphin, 2022-08-23:
> Perhaps we should remove DUB-H4 from supported table completely...

seems so

mvp, 2022-08-24:
Removed from supported list

commit a76f45f3 (2022-08-24): Remove D-Link DUB-H4 rev B from the list

D-Link DUB-H4 rev B does not seem to support VBUS off, remove from supported list.
Closes #440.
closed 2022-08-24 by mvp