mvp/uhubctl #442: Skip querying/printing the port status

real-or-random, 2022-08-23:
Would it possible to skip querying the port status for `on`, `off`, and `cycle`? I think that takes a lot of time with the hubs on my Acer monitor. If yes, do you want me to test this first with a simple hack, e.g., commenting out the relevant call to `print_port_status()` in the code?

(I'll probably later open a PR that adds the monitor to the device list. It mostly works but it has a few weirdnesses and I need to do some further testing...)

real-or-random, 2022-08-24:
Okay, so it turns out that this call is pretty slow (~500 ms), but apparently only for the USB3 hubs:

 https://github.com/mvp/uhubctl/blob/master/uhubctl.c#L391-L399

mvp, 2022-08-24:
It must be an issue with your hub, we should debug it first. But, there is already a flag -N to avoid getting detailed descriptor info which should speed it up considerably especially for misbehaving hubs.

real-or-random, 2022-08-24:
> It must be an issue with your hub, we should debug it first.

Do you have a suggestion how to start?

> But, there is already a flag -N to avoid getting detailed descriptor info which should speed it up considerably especially for misbehaving hubs.

Ah okay, I haven't noticed it because it's not yet released. But I just tested with `master`. The flag won't help in my specific case. The slow call is the one I mentioned above. Getting the detailed descriptor info seems quick with these hubs.

mvp, 2022-08-24:
> Do you have a suggestion how to start?
This call is the basic functionality needed for uhubctl to work, I don't see the point of skipping it.
As for debugging, all that we do here is to ask libusb send low level USB control message to the hub to tell its current state.
Perhaps your USB3 hub is defective? Or needs firmware update? Can you try it with another hub?
I did not see this affecting any other hubs I ever looked at.

real-or-random, 2022-08-24:
Ok yes, there's not much I can do then.

It's the hub in this Acer monitor. https://www.acer.com/ac/en/US/content/support-product/8918?b=1
Firmware is most recent.

It also seems that the hub has strange duality: It shows up as 5-port USB2 hub and 3-port USB3 hub. For example, the port I'm currently using powers off if and only if port 1 on the 5-port USB2 hub is `off` and port 1 on the 3-port USB3 hub is `off`

The only thing is that I could try to figure out if only querying the status is slow or also setting the status is slow. I could this but even then I'm not sure if this would help others.

mvp, 2022-08-25:
Thanks, I will close this issue then since it is not actionable.

closed 2022-08-25 by mvp

real-or-random, 2022-08-26:
Do you think it would make sense to include the monitor/hub in the list in the README? It's definitively supported and I can get VBUS off. Not sure what the policy is but it seems similar to other models with caveats, and we could add a link to this issue as done for some other models.

mvp, 2022-08-26:
Sure, we can add it to the list linking to this issue