mvp/uhubctl #445: Anker PowerExpand+ 7-in-1 USB-C PD Media Hub (Model A8346) (advertised as "Anker 341" on Amazon) partially works (no VBUS off)

jschwartz-cray, 2022-08-26:
Note that the VID:PID of this hub is identical to all the AmazonBasics ones in the README.md, but it is most definitely Anker branded, came in an Anker box, etc.  This is why I was so specific in the title.  I did purchase it from Amazon though.

```
$ uhubctl
Current status for hub 20-3 [2109:2817 VIA Labs, Inc. USB2.0 Hub 000000000, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0000 off
  Port 3: 0100 power
  Port 4: 0100 power
  Port 5: 0503 power highspeed enable connect [2109:8817 VIA Labs, Inc. USB Billboard Device 0000000000000001]
```

Given how difficult it can be to find some of these specific models I would encourage you guys to expand the README.md to include USB hubs that don't work at all as well as ones with partial support (data can be turned off, but not VBUS off, like my hub above).  I was struggling to find a reasonably priced hub from your existing list that didn't have a bunch of features I don't need and was actually in stock somewhere.  I eventually took a chance on this one because of the other Anker hubs you had listed that worked and unfortunately it's not fully supported, but for my use case it works fine.

For people shopping for a hub, lists of those that DO NOT work with uhubctl and those which only partially work is about as useful as the ones that are known to work when so many of them are so old.

mvp, 2022-08-26:
Thanks for your report! But sorry, we do not add hubs which don't support VBUS off. These days majority of new USB3 hubs do report PPPS support, but don't support VBUS. We cannot boil the ocean and list all unsupported hubs.

closed 2022-08-26 by mvp

jschwartz-cray, 2022-08-26:
And it doesn't bother you at all that this "unsupported" hub has the exact same VID:PID as ones that you list as supported in the README.md?  How do you reconcile that?

mvp, 2022-08-26:
There are only handful of USB chipset manufacturers (NEC/Renesas, Via, Realtek, SMC/Microchip, TI), and all of them do support PPPS. By default chipset vendors define some vid:pid. It is up to hub OEM to possibly override vid:pid with their own (very few actually do that), and also provide circuitry to actually support PPPS - not many do that either. If they don't support PPPS, they should disable such advertisement in chip configuration (again, many don't bother or know how to do that either).

So in the end, having identical vid:pid only tells us that hubs are likely to be produced using the same USB chipset vendor, but not much else.