mvp/uhubctl #446: Python library for uhubctl

dakotahorstman, 2022-08-30:
Hello @mvp

I thought I would make you aware that I've developed a Python wrapper for uhubctl. This wrapper is nothing more than a high-level interface to call uhubctl from Python and can be found here: https://pypi.org/project/pyhubctl/

My workplace, DuraTech, uses uhubctl in a couple of our in-house projects, so I thought I'd make you aware that we've now open-sourced parts of our internal wrapper in case you or anyone would like to use it.

My Regards,
Dakota

mvp, 2022-08-31:
Thanks for creating this. However, I'm not sure where and how this should be mentioned in uhubctl. I would expect that there will be multiple uses and wrappers by different frameworks out there: Python, Ruby, Perl, who knows what else. We can't really mention them all?

dakotahorstman, 2022-08-31:
Whether or not you mention this or future wrappers is up to you. Putting them into your "notable projects", or adding a new section like it, may be the best approach. I'm not sure how you compiled your "notable projects", but perhaps you only add a mention of it if a user brings it to your attention.

mvp, 2022-08-31:
I just created notable projects when people mention it and it seemed cool project to be aware of.
Regarding bindings to other languages (including python), you're right that it is better to create separate section in readme.

dakotahorstman, 2022-09-01:
I see! Sounds like a plan then.

I didn't even see you already had a wrapper for Python listed made by someone else. Ah well, now you have two! I'll close this ticket since you've seen it now.

closed 2022-09-01 by dakotahorstman