mvp/uhubctl #447: Fixing USB version for Rosonway hubs

melyux, 2022-08-31:
In the compatibility table, the Rosonway hubs all have USB version "3.2" as their version. Assuming we're using the terminology 2.0 for 480Mbps, 3.0 for 5Gbps, 3.1 for 10Gbps, 3.2 for 20Gbps, and 4 for 40Gbps, these should instead be versioned like this:

* RSH-518C: 3.0
* RSH-A13: 3.1
* RSH-A16: 3.0
* RSH-A104: 3.1

I'm also pretty sure the HP "USB-C Dock G5 5TW10AA" dock that's also listed as 3.2 is only 3.0 (5Gbps).

mvp, 2022-08-31:
These versions were taken from what hub report about themselves in USB descriptors. Unfortunately, I don't have conclusive way to confirm what actual USB version hub should be rated at without extensive testing with different end devices at different speeds.

melyux, 2022-08-31:
The USB terminology is so confused that the "3.2" designator is actually correct for all 3 of the USB 3.x speeds, but not very useful for differentiating between them. Using 3.0, 3.1, 3.2 based on the top speed they support is a lot more useful when looking at the table to decide which one to buy when looking for something with `uhubctl` support.

mvp, 2022-08-31:
Again, I don't mind fixing USB version to what it actually should be. But, I don't know if e.g. RSH-A16 is 3.0, 3.1 or 3.2. I don't have any USB 3.2 device to connect to RSH-A16 and check if it truly supports 20gbps.

melyux, 2022-08-31:
Looking at the Amazon/own product pages for these shows the maximum speed though, so why not look at that? e.g. RSH-A16 doesn't advertise anything higher than USB 3.0 5Gbps, so it's not going to be 20Gbps.

mvp, 2022-08-31:
Hmm. Reading USB 3.2 spec: https://www.usb.org/sites/default/files/usb_3_2_language_product_and_packaging_guidelines_final.pdf
it says:

The USB 3.2 specification absorbed all prior 3.x specifications. USB 3.2 identifies three
transfer rates, USB 3.2 Gen 1 at 5Gbps, USB 3.2 Gen 2 at 10Gbps and USB 3.2 Gen 2x2 at
20Gbps. It is important that vendors clearly communicate the performance signaling that a
product delivers in the product’s packaging, advertising content, and any other marketing
materials.

* USB 3.2 Gen 1
    * Product capability: product signals at 5Gbps
    * Marketing name: SuperSpeed USB
* USB 3.2 Gen 2
    * Product capability: product signals at 10Gbps
    * Marketing name: SuperSpeed USB 10Gbps
* USB 3.2 Gen 2x2
    * Product capability: product signals at 20Gbps
    * Marketing name: SuperSpeed USB 20Gbps

Per this spec, anything USB 3.x is actually USB 3.2.

In other words, per your logic we should have this table:

| USB 3.2 spec       | Old USB version |
| ----------------- | ----------- |
| USB 3.2 Gen 1      | USB 3.0 |
| USB 3.2 Gen 2      | USB 3.1 |
| USB 3.2 Gen 2x2  | USB 3.2 |

mvp, 2022-08-31:
@melyux , I have updated USB versions in README per your suggestion - thank you!

closed 2022-08-31 by mvp

melyux, 2022-11-03:
Hi @mvp, I had another look through the README and found some others that have the wrong USB version per our version scheme. These are all listed as 3.1, but are actually all 3.0 (5Gbps max):

* **Aukey CB-C59** ([only](https://www.digitec.ch/en/s1/product/aukey-cb-c59-usb-c-docking-stations-usb-hub-9029747) claims 3.0)
* **BenQ PD2700U** (has no mention of 10Gbps speeds, implying it's only USB **3.0**)
* **Centech CT-USB4HUB ReTRY HUB** ([only](http://www.centech.jp/?p=625) does 5Gbps)
* **Dell S2719DGF** ([only](https://www.dell.com/en-us/shop/dell-27-gaming-monitor-s2719dgf/apd/210-arcj/monitors-monitor-accessories#techspecs_section) claims 3.0)
* **Delock 62537** ([only](https://www.delock.com/produkt/62537/merkmale.html) claims 3.1 Gen 1 = 3.0 and 5Gbps speeds)
* **ExSys EX-1113HMS** ([only](https://www.reichelt.com/de/en/16-ports-usb-3-0-3-1-metal-hub-exsys-ex-1113-p199309.html?r=1) claims 3.1 Gen 1 = 3.0)
* **LG 27MD5KL-B** ([only](https://9to5mac.com/2017/01/02/lg-ultrafine-5k-display-faq/) [does](https://mobilesyrup.com/2022/03/23/real-world-comparison-apple-studio-display-lg-ultrafine-5k/) USB 3.0)
* **Microchip EVB-USB5807** ([only](https://www.microchip.com/en-us/development-tool/EVB-USB5807) claims 3.1 Gen 1 = 3.0)
* **UUGEAR MEGA4** ([only](https://www.uugear.com/product/mega4-4-port-usb-3-ppps-hub-for-raspberry-pi-4b/) claims 5Gbps = 3.0)

USB 3.1 hubs are still pretty rare.

mvp, 2022-11-03:
Thank you @melyux, device table updated per your suggestions!

commit d75b10d6 (2022-08-31): Fixed USB version for few supported devices to match their spec

Per USB 3.2 spec, any USB 3.x device can claim itself as USB 3.2 per following table:

| Old style USB version | USB 3.2 specification | Speed, gbps |
|-----------------------|-----------------------|-------------|
| USB 3.0               | USB 3.2 Gen 1         | 5           |
| USB 3.1               | USB 3.2 Gen 2         | 10          |
| USB 3.2               | USB 3.2 Gen 2x2       | 20          |

https://www.usb.org/sites/default/files/usb_3_2_language_product_and_packaging_guidelines_final.pdf

And this is what happens - [... cut]