mvp/uhubctl #451: Inateck HB2025A works on USB 2.0 only

ehntoo, 2022-09-02:
I just tested an Inateck HB2025A on my Apple Silicon MacBook running macOS 12.5 and uhubctl version 2.4.0. Per-port power control works perfectly when the hub is connected via a USB 2.0 connection. When connected via USB 3.0, power commands report success but do not switch power off, very similar to #434 which reports the same vid/pid pair.

`uhubctl` output when connected via USB3
```
Current status for hub 0-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OVJ6]
  Port 2: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OK94]
  Port 3: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OO4F]
  Port 4: 0103 power enable connect [0403:6010 zepan@sipeed Sipeed USB to Dual Uart]
```

Attempting to switch ports off - port power remains on:
```
% uhubctl -l 0-1 -a off -p 1
Current status for hub 0-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 0-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
Current status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OVJ6]
Sent power off request
New status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
% uhubctl -l 0-2 -a off -p 1
Current status for hub 0-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
Sent power off request
New status for hub 0-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 00a0 off
Current status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
Sent power off request
New status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
```

`uhubctl` output when connected through a cheap USB 2-only hub:
```
Current status for hub 0-1.1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OVJ6]
  Port 2: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OK94]
  Port 3: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OO4F]
  Port 4: 0103 power enable connect [0403:6010 zepan@sipeed Sipeed USB to Dual Uart]
```

Successfully switching port 1's power off and back on:
```
% uhubctl -l 0-1.1 -a off -p 1
Current status for hub 0-1.1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OVJ6]
Sent power off request
New status for hub 0-1.1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
% uhubctl -l 0-1.1 -a on -p 1
Current status for hub 0-1.1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0000 off
Sent power on request
New status for hub 0-1.1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0101 power connect [0403:6015]
```

Thanks for the project and for maintaining such a nice compatibility database!

ehntoo, 2022-09-02:
An additional partial-compatibility report for the related Inateck HB2025.

I just figured out a way to connect the USB-C connected Inateck HB2025 (non-A) via USB 2.0 and can confirm that the HB2025 with a usb-c plug _also_ will switch power power correctly when connected via USB 2.0 but not via USB 3.0.

Failing to switch port power off when plugged directly into my MacBook:
```
% uhubctl
Current status for hub 0-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OVJ6]
  Port 4: 0100 power
% uhubctl -l 0-1 -a off -p 3
Current status for hub 0-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 3: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 0-2 [2109:0822 VIA Labs, Inc. USB3.1 Hub 000000001, USB 3.20, 4 ports, ppps]
  Port 3: 00a0 off
Current status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 3: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OVJ6]
Sent power off request
New status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 3: 0000 off
```

Successfully turning port power when plugged into a hub with usb-c downstream ports:
```
% uhubctl
Current status for hub 0-1.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OVJ6]
  Port 4: 0100 power
Current status for hub 0-1 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0503 power highspeed enable connect [05e3:0749 Generic USB3.0 Card Reader 000000001532]
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
% uhubctl -l 0-1.4 -a off -p 3
Current status for hub 0-1.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 3: 0103 power enable connect [0403:6015 FTDI FT230X Basic UART DO01OVJ6]
Sent power off request
New status for hub 0-1.4 [2109:2822 VIA Labs, Inc. USB2.0 Hub 000000001, USB 2.10, 4 ports, ppps]
  Port 3: 0000 off
```

mvp, 2022-09-05:
Sorry for belated response, but did you check VBUS off support? We do not add hubs that don't support VBUS off.
That is, does connected phone stop charging when you turn port power off? Since this hub does not have external power supply, it is very unlikely to support it properly. In particular, this line from hub official documentation states makes me extremely skeptical:
> Please note that the hub does not support charging.

Regarding USB2 vs USB3 quirks, did you only see this issue when using Mac? Can you test it on Linux host?

ehntoo, 2022-09-05:
I can confirm that the HB2025 and HB2025A control VBUS on a per-port basis when given the command from uhubctl. VBUS power is removed from the corresponding port when `-a off` is commanded and re-applied after `-a on`. I understand your skepticism, I was very surprised when it worked.

Regarding your question about Linux hosts, I just tried the hubs on a laptop running Arch Linux, and saw the same quirky USB2/USB3 behavior. It does not appear that the USB2-only VBUS switching is a Mac-related issue.

mvp, 2022-09-05:
Thanks for reporting @ehntoo! I added this hub to the list.

closed 2022-09-05 by mvp

pag, 2024-02-04:
And I also bought one after reading this comment, it being one of the few options listed in the readme which is 1) recent, 2) small, and 3) available on amazon in the UK. I can confirm that it works if I connect it to a MacBook Pro (M1) via a simple USB A to C adapter which does not support USB 3. But just a note that of the three ports on the hub, only two supported power switching: the one nearest the cable leaving the port does not.

mvincm, 2026-01-07:
I could confirm. Works for me. Both USB-A and USB-C version of this hub from Amazon.

@pag 
Exactly the same situation. Must force USB2.0 via adapter and "only" port 2 and 3 is ready for VBUS disconnect.

commit a1bc8682 (2022-09-05): Add Inateck HB2025A as compatible device

This works only if connected via USB2.

Closes #451.