mvp/uhubctl #452: Support hub

nioan, 2022-09-05:
I found an ABLEWE hub that supports PPPS. 

The VID:PID is 2109:2817. 
Manufacturer ‏ : ‎ ABLEWE
ASIN ‏ : ‎ B07YWWZJ65
Item model number ‏ : ‎ NU-501

Here's a link on amazon.
https://www.amazon.co.uk/gp/product/B07YWWZJ65/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Consider adding affiliate links to the list of devices. It would be super useful for users, and maybe some very well-earned revenue will come your way :)

mvp, 2022-09-05:
Can you please check if VBUS off is supported by this hub? We do not add hubs which don't support VBUS off.

As for affiliate links and possible revenue, I don't think it will be measurable support for me to bother with.

If someone really wants to help, much better option would be to send me hub samples. E.g. Rosonway sent me RSH-A16 over a year ago which allowed me to significantly improve USB3 support. 

These days I am very interested in checking USB4 hubs like Caldigit Element hub. I expect that it should be supported, but I have no way to tell, and hub itself is quite pricey to buy it just for experimentation.

nioan, 2022-09-05:
No, it doesn't - sorry should have checked first.

mvp, 2022-09-05:
Bummer. Still, thanks for reporting!

closed 2022-09-05 by mvp