mvp/uhubctl #453: Apple  Pro Display XDR works.

mischievous, 2022-09-06:
05ac:9139

mvp, 2022-09-06:
Please check that VBUS off is supported - that is if connected phone stops charging if you turn port off. We do not add devices which don't support VBUS off.

mvp, 2022-09-06:
Thanks. Can you please also provide output of `uhubctl` with this display attached? You can redact any connected devices, but I want to see everything else to see vid:pid, number of ports, USB2/USB3 etc. Thanks!

mischievous, 2022-09-06:
Current status for hub 2-1 [05ac:9139 Apple Inc. USB2.0 Hub C02C307KML9V, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0d8c:0171 C-Media Electronics Inc. REIYIN Audio]
  Port 2: 0503 power highspeed enable connect [0bda:5411 Generic 4-Port USB 2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 3: 0503 power highspeed enable connect [045b:0209, USB 2.00, 4 ports, ppps]
  Port 4: 0103 power enable connect [05ac:9243 Apple Inc. Pro Display XDR C0295040014JLC0A7]

Current status for hub 2-1 [05ac:9139 Apple Inc. USB2.0 Hub C02C307KML9V, USB 2.10, 4 ports, ppps]
  Port 1: 0503 power highspeed enable connect [0d8c:0171 C-Media Electronics Inc. REIYIN Audio]
  Port 2: 0000 off
  Port 3: 0503 power highspeed enable connect [045b:0209, USB 2.00, 4 ports, ppps]
  Port 4: 0103 power enable connect [05ac:9243 Apple Inc. Pro Display XDR C0295040014JLC0A7]

mvp, 2022-09-06:
Thank you. I'm very surprised that it didn't show USB3 items. Are you using USB3 cable? It may not work (or not work correctly) when connected using compliant cable.

mischievous, 2022-09-06:
I have a thunderbolt cable connected to it.  Everything plugged into it right now is usb 2. 

Alexandra

mvp, 2022-09-07:
Thanks, added to the list as USB 2.0 device (odd!).

commit 7099809b (2022-09-07): Add Apple Pro Display XDR as compatible device

Closes #453.
closed 2022-09-07 by mvp