mvp/uhubctl #454: Connecting a hub provides two new hubs with same location (one hub in supported list)

nioan, 2022-09-06:
When connecting RSHTECH/RSH-A104 [4 ports] I'm getting two usb hubs, one with 4 and one with 5 ports. Both have the same location, which means that I get the following error when trying to switch off one port:

Error: changing port state for multiple hubs at once is not supported.

The connection is  RSHTECH/RSH-A104 [4 ports]  --> ABLEWE/NU-501 --> Macbook pro

Is there a work around?

This is the output of uhubctl.
Current status for **hub 20-4** [**2109:2817** VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0503 power highspeed enable connect []
  Port 4: 0503 power highspeed enable connect []
Current status for hub 0-4 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 0-2 [2109:0817 VIA Labs, Inc. USB3.0 Hub, USB 3.10, 4 ports, ppps]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 0203 power 5gbps U0 enable connect []
Current status for hub **20-4** [**2109:2817** VIA Labs, Inc. USB2.0 Hub, USB 2.10, 5 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0503 power highspeed enable connect []
  Port 5: 0503 power highspeed enable connect []

mvp, 2022-09-07:
Make sure you are using uhubctl version 2.4.0 or later. If you are using anything older it may have issues.
If you are on MacOS 12.4 it has USB stack bug, upgrade to MacOS 12.5 - this should fix this problem.
MacOS prior to 12.4 are not affected by this, only 12.4.

nioan, 2022-09-07:
Thank you. MacOS 12.4, so that should be the problem.

closed 2022-09-07 by mvp