mvp/uhubctl #457: Can't toggle power to disk using Linux 6rc3

emsaks, 2022-09-13:
I've been using uhubctl to toggle power to frozen/hung disks. But now, under Linux 6 (rc3) if the disk is opened (e.g. a process is reading it, or in my case, `dmsetup` has a dependency on it) the command will hang until the disk is closed. (I've also tried using the sysfs-disable branch).

mvp, 2022-09-13:
@mgrzeschik, @hnez would know more why it broke in Linux 6 rc3.

emsaks, 2022-09-13:
How can I contact them (or will they be notified by the @ mention)?

mvp, 2022-09-13:
I guess. I recommend to use older Linux kernel for now. Using Linux 6 is living on the edge at the moment.

mgrzeschik, 2022-09-19:
I for now can not see what change could have introduced this behavior. This could also be an issue with some specific peripheral/controller driver. At least I can say that the patch introducing the
/sys/bus/usb/devices/.../<hub_interface>/port<X>/disable sysfs property is not involved in this. :)

mvp, 2022-09-22:
@emsaks, now that #450 is merged, can you try this again - build uhubctl from master branch?

mvp, 2022-09-25:
Closing for inactivity. Please feel free to reopen if you have anything to add @emsaks

closed 2022-09-25 by mvp

mgrzeschik, 2022-09-29:
Did you use the latest version of uhubctl to trigger this. Including the recent changes of #450 that was merged lately?

emsaks, 2022-09-29:
No; I built your tree myself before the merge.

mvp, 2022-09-29:
Can you try building it from master branch and check if it works for you @emsaks?

mvp, 2022-10-12:
Thanks anyway! BTW, official kernel 6.0 was released few days ago.