mvp/uhubctl #467: Feature request: Raspberry Pi Zero W

kmanc, 2022-11-04:
Hey folks! I read through the docs and noticed that a few RPi models are supported but the Zero is not one of them. Is this a technical limitation or a matter of resources? I'm working on a little for-fun project that looks like it could benefit from the use of this tool on a RPi Zero, so if it's doable and there is any way I can help please let me know!

mvp, 2022-11-04:
Unfortunately, Raspberry Pi Zero hardware does not support power switching - this is technically impossible.

However, it is perfectly possible to attach any of listed supported hubs into Raspberry Pi Zero and control power on such hub ports.

closed 2022-11-04 by mvp

kmanc, 2022-11-04:
Understood, thanks :)