mvp/uhubctl #468: Linux permissions section

kmanc, 2022-11-15:
Hey @mvp, really nice work on uhubctl! I'm a new user for a particular project of mine and although I have it working if I have root permissions I seem to be having some trouble with the non-sudo permissions steps listed in the README. Any ideas?

Here's my rules file (except it's in /etc/udev/rules.d and it doesn't have the `.txt` extension
[52-usb.rules.txt](https://github.com/mvp/uhubctl/files/10009992/52-usb.rules.txt)

I ran the command to add my account to the dialout group

Then ran the `udevadm trigger` command.

I also tried restarting the Raspberry Pi (3B rev 1.2) to no avail

mvp, 2022-11-15:
Do you have `dialout` group? Did you add yourself to that group?

If dealing with group membership sounds complicated, you can make permission to be 0666 and remove group from your rule

mvp, 2022-11-15:
Basically, on Linux before 6.0 this one line in rule file should always work:
```
SUBSYSTEM=="usb", DRIVER=="hub", MODE="0666"
```

mvp, 2022-11-15:
Rule provided in default config:
```
SUBSYSTEM=="usb", DRIVER=="hub", MODE="0664", GROUP="dialout"
```
will work only if you have linux group `dialout` (check if there is such group in `/etc/group`), and you are member of it. You can add yourself to this permission group by running:
```
sudo usermod -a -G dialout $USER
```
If you don't like `dialout`, you might want to create group `usb` or `usbaccess` - whatever you like.

mvp, 2022-11-19:
I suppose this answered your question. If it didn't, please reopen this issue.

closed 2022-11-19 by mvp

kmanc, 2022-11-19:
Hey! Apologies I've been traveling.

To sum up, I had actually initially tried the settings that don't involve a special group (so the one with mode "0666") and for whatever reason I couldn't get that to work. Then I tried creating the new group, assigning permission to that, and adding my user to the group. That also didn't work, which I assume means I'm making some dumb mistake somewhere.

Instead I ended up `sudo chmod u+s` the executable, which probably isn't the ideal solution but it gets the job done