mvp/uhubctl #469: Feature request: turn off port

N6REJ, 2022-11-20:
Is it possible to have your program work on a port if no hub is detected?
When using octoprint the 5v from usb port powers up the screens which is a problem.
Would be nice if we could turn the 5v off on the usb port itself.

mvp, 2022-11-20:
USB ports cannot exist by themselves in a thin air - they always belong to some hub. Even if port is onboard, it still belongs to some USB hub, perhaps root hub. Root hub means one where connection to USB bus originates from.

uhubctl can control port power on any USB as long as advertises power control support. You can try operating hubs that don't advertise such support with option -f (--force), but it is extremely unlikely to actually work.

closed 2022-11-20 by mvp

N6REJ, 2022-11-20:
when I type uhubctl it says no hubs exist

N6REJ, 2022-11-20:
apparently I had to use ```sudo uhubctl``` not just ```uhubctl```
it shows 2 hubs with 8 ports.  One port of which says broadcom but I don't know if thats the right port or not.  This is on a laptop