mvp/uhubctl #471: Not an issue, more a note: DRIVER=="usb" for Raspberry Pi 4 (bullseye 11)

RobMeades, 2022-11-26:
First, thanks for this marvelous utility, significantly useful in my case to avoid having to add expensive external HW when a Raspberry Pi is used to control some devices I am testing.

The note is just the the `udev` rule to de`sudo`ise `uhubctl` didn't for me work out of the box: the hubs on my Raspberry Pi 4 show up in `udevadm` as:

```
    SUBSYSTEM=="usb"
    DRIVER=="usb"
```

...so I had to edit the `52-usb.rules` file to change `DRIVER=="hub"` to `DRIVER=="usb"` and then it worked.

This is with `Linux raspberrypi-1 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux`.

mvp, 2022-11-26:
Ugh, it must be related to older Linux kernel versions. I will add a better rule that covers this case too - thank you!

mvp, 2023-01-05:
Fixed examples in 19dfdad.

closed 2023-01-05 by mvp