mvp/uhubctl #473: Only show permissions warning if there are problems with matched devices

silbe, 2022-12-14:
`uhubctl` shows a permission warning if it cannot access one of the hubs on the system even when limited to a certain vendor ID and location and the action was successful:

```
$ uhubctl -n 2109 -l 2-3.4 -p 4 -a off
There were permission problems while accessing USB.
Follow https://git.io/JIB2Z for a fix!
Current status for hub 2-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0107 power suspend enable connect [2357:0604]
Sent power off request
New status for hub 2-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0000 off
$ uhubctl -n 2109 -l 2-3.4 -p 4 -a off > /dev/null && echo successful
There were permission problems while accessing USB.
Follow https://git.io/JIB2Z for a fix!
successful
```

When running `uhubctl` often (e.g. for automated testing) this creates a lot of noise. It would be nice if `uhubctl` would
- only show permission warnings when there were actual problems (action not successful, incomplete listing) and/or
- not try to open hubs that don't match the vendor ID and/or the location.

mvp, 2022-12-14:
Please tell which version of uhubctl you are using. If it is not 2.5.0, please try it first.

mvp, 2022-12-21:
Closing for inactivity. Please reopen if you could reproduce it on uhubctl 2.5.0. Thanks!

closed 2022-12-21 by mvp

silbe, 2022-12-23:
I tried latest `master` (commit 19dfdad7f6c00fa9c5e3b7d926589782300fc20b) and while listing ports (`uhubctl` without parameters) doesn't show a warning, switching on/off a port still does:

```
$ uhubctl -n 2109 -l 2-3.4 -p 4 -a on
There were permission problems while accessing USB.
Follow https://git.io/JIB2Z for a fix!
Current status for hub 2-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0107 power suspend enable connect [2357:0604]
Sent power on request
New status for hub 2-3.4 [2109:2817 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 4: 0107 power suspend enable connect [2357:0604]
```