mvp/uhubctl #474: Solving the "switching itself back on" conundrum

RobMeades, 2022-12-21:
I am using your marvelous code on a test system which consists of multiple Raspberry Pis, each RPi controlling a set of HW under test, driven over USB.

Where the HW under test happens to be drive from a debugger, specifically in the case of Segger JLink Base or ST-Link debug chips, they have that "turn themselves back on" tendency you discuss on the front page.  I've tried setting 1000 retries:

`uhubctl -a 0 -l 1-1 -r 1000`

...but some still persist in staying on, so I was going to try:

```
echo 0 > sudo tee /sys/bus/usb/devices/${location}.${port}/authorized
```

...but I don't fully understand how I to construct the command.  I guess `${location}` is `1-1`, but how do I determine the `${port}` part?  None of the Raspberry Pis know exactly what is connected to them, this all being driven from a generic `Jenkinsfile`, so ideally I need a "hammer" that will work in all cases.

mvp, 2022-12-21:
`uhubctl` without parameters will list all hubs e.g. 1-1 as well as port numbers with devices attached to them (1, 2, ...).

mvp, 2022-12-21:
Hub location for RPi should be always 1-1, and it will always have only 4 ports 1 to 4. If may as well do it for all of them

RobMeades, 2022-12-21:
Ah, yes, got it.  I will do that, thanks!

closed 2022-12-21 by mvp