mvp/uhubctl #479: Device report: D-Link DUB-H4 E1E - on/off only affects fast-charging port 4

mk-fg, 2022-12-30:
Device: D-Link DUB-H4
H/W Version: E1
Part No: ... E1E
Dates: 2021-05-26 (V1.0 Quick Installation Guide), 2022-12-30 (purchase).
uhubctl-2.5.0 id: 2109:2210 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps
Status: power switching only works on "fast charging" port 4, changing it on ports 1-3 has no effect.

Tested actual power status on each port with a dollar-store as-dumb-as-they-come USB lamp.
Tried plugging it into USB 2.0 ports from mobo header and 3.0 - power controls work the same in both.
uhubctl version 2.5.0 (2022-12-30).

Having microUSB power adapter plugged into hub does not seem to change PPPS effects in any way.
Plugging microUSB power in seem to reset the hub with a usb-lamp blink and USB re-init.
Unplugging USB 2.0 with external power does not seem to blink usb-lamp, but resets the PPPS status of all ports (i.e. enables port 4 power if it was disabled).

Does not seem to be same result as what was reported for H4/E earlier in #212, likely because of "E1" hardware.
Note that body color is also not black - seem to be about #808080 (under LED lights) matte powder-coat metallic finish.
(afaict blueish/purple hue in images below is a color cast/balance bug in the crap-phone camera/processing)

[image]
[image]

uhubctl:
```
Current status for hub 3-1 [2109:2210 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports, ppps]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

lsusb -v:
```
Bus 001 Device 007: ID 2109:2210 VIA Labs, Inc. Hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass            9 Hub
  bDeviceSubClass         0
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x2109 VIA Labs, Inc.
  idProduct          0x2210 Hub
  bcdDevice            7.25
  iManufacturer           1 VIA Labs, Inc.
  iProduct                2 USB2.0 Hub
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0019
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0001  1x 1 bytes
        bInterval              12
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x002a
  bNumDeviceCaps          3
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat           4 micro seconds
    bU2DevExitLat         231 micro seconds
  Container ID Device Capability:
    bLength                20
    bDescriptorType        16
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {30eef35c-07d5-2549-b001-802d79434c30}
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             4
  wHubCharacteristic 0x00e9
    Per-port power switching
    Per-port overcurrent protection
    TT think time 32 FS bits
    Port indicators
  bPwrOn2PwrGood      175 * 2 milli seconds
  bHubContrCurrent    100 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0000
   Port 2: 0000.0000
   Port 3: 0000.0000
   Port 4: 0000.0000
Device Status:     0x0001
  Self Powered
```

One port switching is pretty much all I needed, so good enough I guess.

Not sure if it's bad/non-spec USB3 extender cable that I've been using, but whole hub seem to reboot and re-negotiate USB ("USB disconnect" + "new high-speed USB device ...") when anything (even no-data-pins lamp) plugged in/out of it (except for uhubctl-powered-off port 4), when I tested it with USB3, which didn't seem to happen in USB2 even once - weird, maybe not great device overall.

Thanks.

mvp, 2023-01-05:
Given list of problems you have encountered, I would rather not add this device to supported list. Thanks for reporting anyway!

closed 2023-01-05 by mvp

mk-fg, 2023-01-05:
Might be worth mentioning in "not working" list, as I got it hoping that it might work similar to "E", which is mentioned as working.
Otoh guess if anyone bothers to google exact hw/pn, they might now find this issue and get info that way, so idk.